JoelBender/bacpypes #5: Invalid reference to register_object_types

JoelBender, 2015-07-31:
In LocalDeviceObject.__init__() there is new code that checks to make sure that the class is registered correctly with its vendor identifier.  When the code was changed to clean up imports, the object module is no longer completely imported, but the reference to object.register_object_type() and object.register_object_types is no longer correct.

ChristianTremblay, 2015-07-31:
In previous version vendorId was vendorID (cap D)... is this changed everywhere ?

JoelBender, 2015-08-01:
No, the Confirmed Private Transfer Request/Ack/Error uses *vendorID*, the classes in basetypes use *vendorId*, and there is a *vendorIdentifier* property of a Device Object.  That hasn't changed anywhere.

JoelBender, 2015-08-03:
I'm assuming this is fixed.  I'm closing the issue so I don't keep working on the issue 5 branch!

closed 2015-08-03 by JoelBender