JoelBender/bacpypes #7: py34 : WARNING:bacpypes.bvllservice.UDPMultiplexer:unsupported message

ChristianTremblay, 2015-07-31:
Message are correctly sent to the network (verified with Wireshark)
Response is misinterpreted somewhere and give the warning...
Not data in the response

JoelBender, 2015-08-01:
This is actually issue #7, I removed the issue name from the title.

JoelBender, 2015-08-01:
The message byte should have been unpacked as unsigned, fixed in the issue-7 branch so far.  To be continued.

JoelBender, 2015-08-03:
I added test code in the **tests/test_pdu/test_address.py** file that pushes the edges of `LocalStation`, `LocalBroadcast`, and `GlobalBroadcast` a little to get the in swing of writing tests.  I suspect that I will have to abandon py25 testing and stick with py27 and py34, the `with self.assertRaises()` isn't available because the `with` statement is in py26.

Note that I added it to the *issue-5* branch by mistake, groan.  *Issue-5* has been merged into *issue-7* and is waiting for some testing before being merged into *master*.

JoelBender, 2015-08-03:
If you have done some testing, pull a new copy of the *issue-7* branch, I improved some error checking and added a boat load of tests.

JoelBender, 2015-08-13:
This has been merged into *stage*

closed 2015-08-13 by JoelBender