JoelBender/bacpypes #9: issue-7

ChristianTremblay, 2015-08-01:
mstp mac address are received as string like : b'\x04' instead of a simple conversion

example : 
```
> whois 3:4
> pduSource = <RemoteStation 3:b'\x04'>
iAmDeviceIdentifier = ('device', 5004)
maxAPDULengthAccepted = 480
segmentationSupported = segmentedBoth
vendorID = 5
```

JoelBender, 2015-08-01:
This is actually issue #9, so I updated the title.

JoelBender, 2015-08-01:
Check out the issue-9 branch.  This is only a problem with py34, but was also a problem with `LocalStation` addresses.  If you can confirm it's fixed, I'll merge this with the master branch.  This is exactly the kind of problem that good unit tests would fix!

ChristianTremblay, 2015-08-01:
I think it's ok 
```
> whois
> pduSource = <Address 192.168.210.253>
iAmDeviceIdentifier = ('device', 210253)
maxAPDULengthAccepted = 1476
segmentationSupported = noSegmentation
vendorID = 245
pduSource = <RemoteStation 2:5>
iAmDeviceIdentifier = ('device', 5)
maxAPDULengthAccepted = 480
segmentationSupported = segmentedTransmit
> exit
Exiting...
```

ChristianTremblay, 2015-08-01:
Regarding unit test, we could think about reference PDUs that we would always create in our unittest... and make all the tests based on that "fake" PDUs

I'm thinking of complete ones similar to what we could receive from the network.

One MSTP, one IP... for example

JoelBender, 2015-08-03:
Setting up a testing framework is going to be a challenge.  I was expecting that the virtual network implementation in the **vlan.py** was going to be sufficient for the stack, and probably will be where the client and server are in the same application.  The ultimate goal is for testing to match the *ASHRAE Standard 135.1 Method of Test for Conformance to BACnet* standard and could be applied to real devices.  It seems like the client and server tests would be identical, they're all just sequences of PDU's.

:-)

I'm closing this issue before the thread gets too off-track.

closed 2015-08-03 by JoelBender