JoelBender/bacpypes #41: Sample Application: ReadPropertyAny.py

Wiff3r, 2015-08-12:
A sample application is needed to show how to read atomic properties from proprietary objects.

Wiff3r, 2015-08-12:
I looked at ReadProperty.py and below is what I think needs to be done, but I don't know how to get the console to accept numbers for the objectType.

In confirmation():
            #if not datatype:
            #    raise TypeError, "unknown datatype"

            if datatype:
                # special case for array parts, others are managed by cast_out
                if issubclass(datatype, Array) and (apdu.propertyArrayIndex is not None):
                    if apdu.propertyArrayIndex == 0:
                        value = apdu.propertyValue.cast_out(Unsigned)
                    else:
                        value = apdu.propertyValue.cast_out(datatype.subtype)
                else:
                    value = apdu.propertyValue.cast_out(datatype)
            else:
                try:
                    tagClass = Tag._app_tag_class[apdu.propertyValue.tagList[0].tagNumber]()
                    tagClass.decode(apdu.propertyValue.tagList[0])
                    value = tagClass.value
                except:
                    raise TypeError, "unknown datatype"

JoelBender, 2015-08-26:
I added the new application `ReadPropertyAny.py` to the `samples` directory available in the `samples` branch, which also includes most of the SourceForge sample applications but not all of them have been carefully examined yet.

The `samples` branch is ahead of the `master` with many bug fixes.  So before running the sample, please do the `python setup.py build install` dance *after* switching branches!  This will get rolled out into the `master` branch during the next release.

I'm closing this issue for now, feel free to reopen it if you have problems.

closed 2015-08-26 by JoelBender