JoelBender/bacpypes #43: Move Documentation

JoelBender, 2015-08-13:
Move the Sphinx documentation from SourceForge.

JoelBender, 2015-08-13:
This is going to be put in a persistent **doc** branch.  Similar to **samples** branch, I want to be able to update the documentation regularly.  Of course when there is a new release, the release notes will be updated, and then the **doc** branch will be merged into the master.  I don't expect that it will be important enough to merge the doc branch into staging, which isn't to say that there won't be documentation changes that are a necessary part of closing issues.

ChristianTremblay, 2015-08-13:
I think we should use the gh-pages branch for doc as it will auto-populate the Web page.

JoelBender, 2015-08-13:
Ah!  I didn't see your idea.  Well, that's something to consider, but the current documentation is already in reStructuredText, and it synchronizes nicely with ReadTheDocs and gets its own URL: http://bacpypes.readthedocs.org/en/latest/

closed 2015-08-13 by JoelBender