JoelBender/bacpypes #48: Date init string parsing is broken

JoelBender, 2015-08-20:
Similar to the Time bug #47 the Date also has a bug, this fails:

    value = Date("1/2/3")

This should be 02-JAN-2003 and correctly calculate the day-of-week.  Trivial fix.

ChristianTremblay, 2015-08-25:
I'll give a try

JoelBender, 2015-08-25:
Thank you!  I just beat on #47 and included tests in the `issue-47` branch, then merged it into `stage`.

ChristianTremblay, 2015-08-30:
Ok... I made a few tweeks as the format mmddyy is not the only one out there (yeah... I'm canadian... and yeah... I speak french... and we never use this format)

So I tried to figure out how I could handle different formats and recognize them... I made a few regex and came back with this result : 

Str : Date
1/2/3 : Date(2/1/1903 )
1/2/3 : Date(2/1/1903 )
1/2/2003 : Date(2/1/2003 )
1/20/2003 : Date(1/20/2003 )
01/20/2003 : Date(1/20/2003 )
11/12/2003 : Date(12/11/2003 )
30/1/2000 : Date(1/30/2000 )
30/1/1230 : Date(1/30/1230 )
30/1/98 : Date(1/30/1998 )
2015/8/30 : Date(8/30/2015 )
2015/08/30 : Date(8/30/2015 )

Whick is pretty close to what I want.

The first one ('1/2/3') is tricky... when do we make the line between 1900 and 2000.... I mean...98 is probably 1998... but 1.... should probably be 2001... 

As for the __str__()....
I think we should stick to a ISO format like yyyymmdd

What do you think ?

JoelBender, 2015-09-01:
Ok, this has been merged into branch `issue-38` and I'm going to start beating on it.  The only thing that strikes me right way is that it is missing the day-of-week pattern I had as the fourth element.  It also seems to be re-defining the year to be the four digit year but the standard has the first element as the year - 1900.

JoelBender, 2015-09-01:
Just to follow up, the pattern matching should allow for some special values, from page 673 of the 2012 standard:

    Date ::= [APPLICATION 10] OCTET STRING (SIZE(4)) -- see 20.2.12
    -- first octet     year minus 1900    X'FF' = unspecified
    -- second octet    month (1..14)      1 = January
                                          13 = odd months
                                          14 = even months
                                          X'FF' = unspecified
    -- third octet     day of month (1..34)
                                          32 = last day of month
                                          33 = odd days of month
                                          34 = even days of month
                                          X'FF' = unspecified
    -- fourth octet    day of week (1..7)
                                          1 = Monday
                                          7 = Sunday
                                          X'FF' = unspecified

No doubt there are many combinations of these special values that are meaningless, and some that are disallowed in some context but allowed in others, and some that are going to be ambiguous.

As for using https://en.wikipedia.org/wiki/ISO_8601, since it doesn't support wildcard characters or day-of-week, that won't work well.  But making **str()** be more universal like "yyyy-mm-dd dow", I'm all for that.

Humm... `"2015-odd-last *"` is the last day of odd months in the year 2015?  Eeep!

ChristianTremblay, 2015-09-01:
I was not aware of those special cases but I'm sure we can add them pretty easily. (Yeah I should have read the standard more closely...)

As for the dow pattern... I don't understand because I used your logic to determine it... 
Are you referring to the call to the function to find the dow if given a year, month, date ?

JoelBender, 2015-09-02:
No, the original pattern had a fourth section of the regular expression.  None of your patterns have it, so the last element of the `date_group` list doesn't get touched and is always `None`, so the `tup_list` always has 255 tacked on the end.  With the other special values from the standard, the `CalcDayOfWeek` should be smarter about recognizing them.

ChristianTremblay, 2015-09-02:
OK ! I understand ! You are right, the pattern doesn't handle the dow

I will send you a patch file instead of pushing to issue-38 to propose some modification to date handling.

Concerning the DOW patterns... do you think we could add the portion of dow you used ?

JoelBender, 2015-09-02:
The more I think about it, the more concerned I get about bloating the primitive with parsing problems.  I added the simple parsing to solve a specific case of generating `WritePropertyRequest` sample statements where the property was a `Time` or `Date`.  Time was so simple, throwing a date regular expression seemed like a no brainer.  But of course they were broken :-/.

How do you feel about removing all the issues with region centric versions like "mm/dd/yyyy" and going with just "yyyy-mm-dd dow"?  The more versions of strings that I matched with your patterns the more conflicts I got, which is to say the same input string could be matched by multiple patterns.  That made it harder to understand if the tests were correct because I was trying to understand your test intent.  And it would make it harder to document.

I say we "punt".  The burden of supporting additional patterns that would be important for some application would be relegated to the BACpypes user (web strings, ISO strings, Javascript date strings,  and of course `strptime()`).

Regarding keyword matching like day-of-week names, we could add month names and `last`, `odd` and `even` as special keywords, but are we overstepping in a different direction?  So match numbers and `*` and that's it?

ChristianTremblay, 2015-09-02:
It's better to reset now than later. We now know that dealing with all sort of date is really hard... and maybe not mandatory for bacpypes.

I think you are right about an important point. If we define bacpypes dates to be yyyy-mm-dd dow... a client app using bacpypes will have to deal with other formats... and that is perfect that way I guess. Keep it simple they say.

If I follow you, the pattern would look only for numbers and * (which is much more simple).
Then if the user wants to use last/odd/even.... he'll know what number to use.

I like that

What do you think of letting the str function output show last/odd/even ? It would be a way to "prove" bacpypes understood the meaning of a 13th or 14th month (or day > 31)

On my side, I'm very comfortable with this view. And it will have been a really good way to dive into regex patterns... even if we don't keep them :-)

JoelBender, 2015-09-04:
I'm not afraid to make leaps from the sublime to the ridiculous, and so in this commit c89ffbc61f8cf1e3d872dcd45f67ba94be944cbb I've added `samples/date_alternative.py` which contains your code horribly twisted (well, the patterns deconstructed and then rebuilt) along with keyword matching and updating the output of the `str()` function as you suggested.

I also added a bit of interesting code that rejects ambiguous strings (that match more than one pattern) but accepts those that end up being unambiguous (because the essential elements of the patterns are identical).  So, for example, '1/1/15' is ambiguous (because it matches `mmddyy` and `ddmmyy`) but is accepted.

To limit the acceptable input patterns to `yyyymmdd` is as simple as commenting out the other variations in the pattern list.  So why do the rest?  It's fun!

ChristianTremblay, 2015-09-04:
The fun part is that I can learn to code... and lean new idioms at the same time :)

I must say I like that. It's easy to read and understand. I like the part of the match to find ambiguous cases. Raising an exception is a lot better than making a choice for the user (as I did). The use of zip is very interesting.

The merge function for regex is something I'll keep in mind for future cases. I really like the way it clarify things.

Do you see any bugs or weirdness that would prevent us to use this as the "official" date class in the future ?

JoelBender, 2015-09-06:
Under the guise of the [robustness principle](https://en.wikipedia.org/wiki/Robustness_principle) I'm moving forward with putting this new code it.  Because the branch was relatively old and very different from the other updates that I've done in #38, this turned out to be a lot more work than I expected and you'll notice that the timeline commits bounce back and forth between the two issue branches.  Yuck!

The end result is that I'm calling this issue closed, and I would like to recall my statement about being a "trivial fix" :-).

closed 2015-09-06 by JoelBender