JoelBender/bacpypes #56: CharacterString Encoding

ChristianTremblay, 2015-09-01:
ERROR:bacpypes.comm.ApplicationServiceElement:exception: TypeError("Can't convert 'bytes' object to str implicitly",)
Traceback (most recent call last):
  File "C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\bacpypes-0.13.1-py3.4.egg\bacpypes\app.py", line 231, in indication
    helperFn(apdu)
  File "C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\bacpypes-0.13.1-py3.4.egg\bacpypes\app.py", line 327, in do_ReadPropertyRequest
    resp.propertyValue.cast_in(value)
  File "C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\bacpypes-0.13.1-py3.4.egg\bacpypes\constructeddata.py", line 1013, in cast_in
    element.encode(tag)
  File "C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\bacpypes-0.13.1-py3.4.egg\bacpypes\primitivedata.py", line 817, in encode
    tag.set_app_data(Tag.characterStringAppTag, (chr(self.strEncoding)+self.strValue.encode('latin-1')))
TypeError: Can't convert 'bytes' object to str implicitly

JoelBender, 2015-09-02:
This has been fixed 00a74fb8a20b2099c9e891aa6eb17088dcf04c9f and is merged into `stage`.  If you are working on `issue-48` I noticed that it is behind `stage`, you might want to merge it into the `issue-48` branch and reinstall, that should give you the other fixes as well.

ChristianTremblay, 2015-09-02:
That's no big deal for now, glad to see it's been fixed. I'll close this

closed 2015-09-02 by ChristianTremblay