JoelBender/bacpypes #57: Special date values

JoelBender, 2015-09-08:
The `Date` class checks to make sure that the month, day, and day-of-week values are not zero.  However, the standard may use the special four-tuple `(0, 0, 0, 0)` or some other special value to signify the last item in a list or an item that should be skipped.  Check the standard for special date values and update the primitive data `Date` code appropriately, or simply close the issue if there aren't any.

ChristianTremblay, 2015-09-11:
Trivial fix :-)

JoelBender, 2015-11-04:
I went through the standard and looked at the places where there are dates (like BACnetDailySchedule, BACnetSpecialEvent, and BACnetDateRange) and I couldn't find any places where all zeros are used.

closed 2015-11-04 by JoelBender