JoelBender/bacpypes #58: Abort and Reject Exceptions

JoelBender, 2015-09-12:
In the `errors` module create an `AbortException` class that is a derived class of `RuntimeError`, then create a series of subclasses such as `APDUTooLong`, `ApplicationExceededReplyTime`, `BufferOverflow`, and the others.  When any of these exceptions are raised during the interpretation of an APDU then trap it and return an `AbortPDU` to the client with the appropriate reason code.

Similarly, create a `RejectException` class with subclass for `InconsistentParameters`, `InvalidParameterDataType`, `InvalidTag`, and the others.  Trap these and return a `RejectPDU` to the client.

Note that BUFFER_OVERFLOW and OTHER are identical names, so call these `AbortBufferOverflow` and `RejectBufferOverflow`.  Putting the Abort and Reject prefix in front of all of the exceptions is superfluous.

JoelBender, 2015-09-26:
I committed code that raises the new `InvalidTag` exception 3c3ad2fe7f1b63d1666a53ced17b09a89c79b0fc and adjusted the tests accordingly 60cba9c9497a6de201b4ef0325f91761930c1f42.  Before I close this issue and merge this branch back into stage I need to test it with py34.

JoelBender, 2015-09-29:
Raising the correct reject exception during decoding is going to be tricky.  I'm merging this into stage as of 5e9f5ea693b4a011fae2f94f94a0d271213ab4bb so it gets into the next release, but leaving the issue open for a while longer.

ChristianTremblay, 2015-09-29:
If I understand correctly, the check is done based on "len(tag.tagData)" which will cover more errors than only a empty bytes right ? (my proposal was only looking for empty bytes)

JoelBender, 2016-05-01:
@ChristianTremblay There are lots of additional checks, not just on the length of the tag data.  Most of the new code behavior should be reflected in the new tests that are included. I remember being pretty happy with it back in February, but I need to brush the dust off (or scrape it off, depending on your viewpoint :smile:) and confirm it before merging it into `stage` for the next release.

ChristianTremblay, 2016-05-01:
I'll have to dig into... not sure I remember everything :-)

JoelBender, 2016-05-06:
I'm going to suspend this issue until I get the tests moved to `py.test`, issue #80.  I want to be able to easily build lots of malformed APDUs and run them through the stack, so I'll need to set-up and tear-down lots of combinations of stacks.  If moving away from `nose` is going to be done, it should be sooner rather than later.

JoelBender, 2016-08-22:
Closing this as it stands, more testing would be nice.

closed 2016-08-22 by JoelBender