JoelBender/bacpypes #59: UDPMultiplexer wrong handling of local broadcast address

ChristianTremblay, 2015-09-12:
I've started an application using a bacpypes.pdu.Address object created from a string ('192.168.210.63/24')

When trying to start the app, this piece of code tries to bind the broadcast address which lead to a windows exception

I commented the lines that create the exception, and just defined self.broadcast = None as a test (which actually work... !?!

bvllservice.py lines 92@99
```
        # create and bind the broadcast address
#        if specialBroadcast and (not noBroadcast):
#            self.broadcast = _MultiplexClient(self)
#            self.broadcastPort = UDPDirector(self.addrBroadcastTuple, reuse=True)
#            bind(self.direct, self.broadcastPort)
#        else:
#            self.broadcast = None
        self.broadcast = None
```
Error : [WinError 10049] The requested address is not valid in its context

Actually, if I comment those line and create the app using the Address object, broadcast messages are sent to 192.168.210.255 instead of 255.255.255.255. This is what I want.

Now, I want to understand what is the goal of this piece of code, why it's here... and why it works when I remove this part :)

JoelBender, 2015-09-13:
This is a windows socket library problem, so I'm poking around in the dark here.  According to [this question](http://stackoverflow.com/questions/4657347/socket-error-ip-address-not-valid-in-its-context-python):

> That's an error Windows gives when you're trying to bind to an address on the local machine that's not assigned to any of the adapters on the machine. If ipconfig doesn't show it, you can't bind to it.

So you can't bind to the LAN broadcast address, but the question is, can you receive packets sent to this address?  My guess is that Windows expects you to want to receive broadcast packets, and it may recognize both LAN broadcast and generic 255.255.255.255 broadcast packets.  This is not the same for datagram packets in Linux.  You'll have to test that theory on a LAN with two Windows machines.

The `broadcastPort` is only used to listen for broadcast messages, it's not used for sending them.  The destination for a broadcast is calculated from the address, so that part works independently of having this special socket.

Someplace I've got some sample code kicking around that will make this easy to test, I'll create a `issue-59` branch and add it.

JoelBender, 2015-09-13:
Ah!  You already created one :-)

ChristianTremblay, 2015-09-13:
I just tested it with 2 machines running Windows. And I don't need to bind the broadcast address : 
See wireshark result : 

[image]

I guess we'll need to check for OS in the code and exclude this part on Windows.

ChristianTremblay, 2015-09-13:
See I'm running two different versions of BAC0 as one of the PC answers with 2 IAm() :)

JoelBender, 2015-09-14:
I just committed f1c8d09ef154f019f0e21e6deb7309b9eec71a30 with a sample application for you to run.  I haven't tried it in Python 3, but otherwise it behaves as expected.  Pay particular attention to the application not recognizing its own broadcasts (which can can get it into trouble if it receives its own I-Am messages and thinks there is an impostor), and not recognizing a message as a broadcast (which means it can't reject confirmed services sent as a broadcast message).

The duplicate I-Am messages is a BAC0 bug, not a BACpypes bug ("Need a hint" email thread).

JoelBender, 2015-09-14:
I tested this code with py34 on Linux with py27 on MacOS and they both exhibit the same characteristics.  Note that I pushed 482f4ec4a670b84bed88dbfd918b6b3480ace1ca which has some updates on UTF-8 encoding, and 46b9b1e1ab1236534d56afadf37d18308bf3b549 which has some debugging updates and more UTF-8 encoding.

ChristianTremblay, 2015-09-29:
At first I get this error

![uploads/9d79553a-ad99-4350-87a1-244e20e7e57e/UDPConsole.JPG](https://s3-us-west-2.amazonaws.com/prod.huboard.com/uploads%2F9d79553a-ad99-4350-87a1-244e20e7e57e%2FUDPConsole.JPG

JoelBender, 2015-09-30:
The [`asyncore.file_dispatcher`](https://docs.python.org/3.4/library/asyncore.html#asyncore.file_wrapper) is only available in Unix.  I pushed up a patch b1ed35395073e22999bd2264bd4a98769f25f53a which will allow you to get passed the error.  I'm not sure that `ConsoleClient` and `ConsoleServer` classes will work with Windows.

ChristianTremblay, 2015-09-30:
Yep... error get passed... but I can't write anything from the console... I'll check if I can find something to bypass and test...

ChristianTremblay, 2015-10-17:
I pushed a modification to bvllservice that will set broadcast = None for Windows platform. 
I then use a complete bacpypes.Address variable for local IP address (that contains the broadcast adress) and it works.
This modification causes no harm to Linux platform

ChristianTremblay, 2015-10-21:
I found out this topic on stackoverflow and I think the answer I'm looking for is in it.
http://stackoverflow.com/questions/14222134/winsock-bind-failing-with-wsaeaddrnotavail-for-directed-broadcast-address

[in windows]
```You should not bind() to a broadcast IP address. You need to bind() to an individual network adapter IP instead. If you want to send out a broadcast message, you bind() to the adapter that is going to send the broadcast, and then sendto() the broadcast IP. If you want to receive a broadcast message, you bind() to the specific adapter whose IP matches the broadcast IP being sent to.```

I've made some test using debug output in BAC0... it was easier to implement the UDPConsole.py logic in my script, instead of trying to build a "console" app on Windows, as file_dispatcher don't work.

Now there's something I don't understand correctly­. The part where I must check if the app can recognize its own broadcast.... There something that tells me that Windows take care of that... but I'm not good enough to prove it :)

```Maybe I can get by with a little help from my friend```

JoelBender, 2015-10-28:
You quoted:

    If you want to receive a broadcast message, you bind() to the specific adapter whose
    IP matches the broadcast IP being sent to.

I have not been able to find a Windows Python example that binds to an adapter, they all seem to bind to an address.  I found [ForceBindIP](http://old.r1ch.net/stuff/forcebindip/) which will force it, but that seems excessive.

ChristianTremblay, 2015-10-28:
the point is, I think that I don't need to bind the broadcast address in
Windows... I just bind to the address of the adaptor I want to use... Maybe
just a bad way to say what's needed...

The fact is that bacpypes works in Windows if I just don't bind the
broadcast address.
I just cannot be 100% sure that it will not see its own broadcast and
recognize it.
In fact, I'm 90% sure it does recognize it as I never saw a IAm() request
from self, being treated as a fantom device in the discovered devices
list... a false positive.

That's why I (for the moment) suggest to "not bind" the broadcast address
for Windows platform...

What do you think ?

JoelBender, 2015-10-28:
Do you think Windows supports SO_BINDTODEVICE?  There is some sample code [here](http://nullege.com/codes/show/src%40p%40y%40pytomation-HEAD%40pytomation%40utility%40miranda.py/129/IN.SO_BINDTODEVICE/python) that leads me to believe that you could bind to the INADDR_ANY address and then set the socket option to the specific interface.  The acid test is when you can (a) open a pair of sockets on the same adapter on different ports, or (b) open a pair of sockets on different adapters on the same port, and in both cases send and receive well formed broadcasts and know where they are going.

Are you satisfied with the current state of the `issue-59` branch?  If you give me a thumbs up, then I'll close the issue, merge it into `stage`, and it's about time for a release!

ChristianTremblay, 2015-10-28:
I tried to run the script (after converting it to Python 3 :-) 
It depends on a module I can't find (import IN)...I'll stop here.

For now, I'm satisfied with issue-59 as it allows bacpypes to run on Windows, being able to use global broadcast AND local broadcast.

This issue can be merged/closed

closed 2015-10-29 by JoelBender