JoelBender/bacpypes #62: Change of Value Reporting

JoelBender, 2015-10-17:
Change of Value Reporting is described in Clause 13.1, along with the descriptions of the associated services in clauses 13.6, 13.7, 13.14, and 13.15.  This issue is to support the object types and services.

mebrunet, 2016-02-06:
I'll chirp in saying that this would be great to have.

JoelBender, 2016-02-21:
Note to those watching this issue (@mebrunet I don't know if you seen this somewhere else) Some changes to the library code have been made and are released in 0.13.7, but the code that uses it remains in `samples/COVMixin.py` for the server side and `samples/SubscribeCOV.py` for the client side.  This issue is staying open because there are some special criteria for `AccessDoorObject`, `LoadControlObject` and `PulseConverterObject` that hasn't been written yet.

This does *not* cover any of the functionality for the _SubscribeCOVProperty Service_, which is going to need a different collection of mix-ins and sample code.  :-\

ChristianTremblay, 2016-11-19:
@JoelBender is this working in 0.15 (issue #87) ? 
Can we close ?

JoelBender, 2016-12-01:
Moved into `stage` and ready to be released.

closed 2016-12-01 by JoelBender