JoelBender/bacpypes #68: Broadcast IP not bound to in OS X

bf1967, 2016-02-03:
The broadcast address is not bound to in OS X.

In bvllservice.UDPMultiplexer.__init__ there is a check to bind to the broadcast address for non-windows machines. Part of that check is "'win' not in sys.platform". OS X reports back with a platform of 'darwin' and thus fails the check and doesn't end up bound to the broadcast address.

ChristianTremblay, 2016-02-04:
Not sure about how you say it...
The check is there to "exclude" Windows machine from binding to the broadcast address.
So, if running on Linux or OSx, the function will try to bind the broadcast address.

That said, on windows, I was getting an error when trying to bind to the broadcast address. Do you get an error on OSx ?

bf1967, 2016-02-04:
I added an exclusion for OS X in the affected module and submitted a pull request #69

JoelBender, 2016-02-04:
I think what we really want is the same kind of check like I did in `task.py`, so how about changing the check to be more pro-active?  Like this:

```
        if specialBroadcast and (not noBroadcast) and sys.platform.startswith(('linux', 'darwin')):
            self.broadcast = _MultiplexClient(self)
            self.broadcastPort = UDPDirector(self.addrBroadcastTuple, reuse=True)
            bind(self.direct, self.broadcastPort)
```

bf1967, 2016-02-04:
Is it only the unix-derivatives this check applies to?  There are 8 possible platforms that could be reported.

https://docs.python.org/2/library/sys.html

I'm not sure if BACpypes gets used in any if the others.  Otherwise I'd agree with the change you've proposed.

JoelBender, 2016-02-04:
Should there be a check for "supported platforms" in the `__init__.py` file and if it's not one that we, the BACpypes community, recognize then we punt with a warning?  I don't know how the other platforms get broadcast packets to the appropriate socket.

bf1967, 2016-02-04:
That would seem to be the friendliest thing to do.  Otherwise someone starts using the package assuming it works on their system of choice.   It becomes very clear which are supported/tested.

JoelBender, 2016-02-04:
Thank you for the reference, I just created an `issue-68` branch.  Please look it over and see if it's acceptable.  Note that the code for Python 3.4 diverges from Python 2.5 and 2.7 slightly because the values for `sys.platform` changed in Python 3.3.

bf1967, 2016-02-04:
Looks good to me.   I closed my pull request by the way.

JoelBender, 2016-02-21:
This has been merged and released in 0.13.7

closed 2016-02-21 by JoelBender