JoelBender/bacpypes #76: Debugging to a rotating file handler

JoelBender, 2016-04-13:
The current `--debug` command line option takes a list of named debugging access points and attaches a `StreamHandler` which sends the output to `sys.stderr`.  I would like an additional way to send the debugging output to a `RotatingFileHandler` by providing a file name, and optionally `maxBytes` and `backupCount`.  For example, this invocation sends the main application debugging to standard error and the debugging output of the `bacpypes.udp` module to the `traffic.txt` file:

    $ python some_application --debug __main__ bacpypes.udp:traffic.txt

By default the `maxBytes` is zero so there is no rotating file, but it can be provided, for example this limits the file size to 1MB:

    $ python some_application --debug __main__ bacpypes.udp:traffic.txt:1048576

If `maxBytes` is provided, then by default the `backupCount` is 10, but it can also be specified, so this limits the output to one hundred files:

    $ python some_application --debug __main__ bacpypes.udp:traffic.txt:1048576:100

So the definition of DEBUG becomes:

    positional arguments:
        --debug [DEBUG [ DEBUG ... ]]
            DEBUG ::= debugger [ : fileName [ : maxBytes [ : backupCount ]]]

The justification is that I quite often run the same application many times with multiple debugging options, first adding more debugging to understand more about what an application is doing and then removing them to temper the volume of debugging output.  By redirecting debugging output to a file, I can limit the amount of output to the screen and dig into the details by looking at the output files when needed.  I also would like the option for other BACpypes to run their applications and send me just one particular layer of debugging for their application without needing them to send everything.

JoelBender, 2016-08-25:
This now works, with an extra flare.  The `maxByets

ChristianTremblay, 2016-11-19:
I think it is safe to close as it works in issue #87 and will be part of 0.15

closed 2016-11-19 by ChristianTremblay