JoelBender/bacpypes #77: Multiple bacnet devices with same ip address

BalazsHoranyi, 2016-04-22:
I am running into issues reading properties off multiple devices that have the same IP address. 
I have 3 subnets each with 20-100 devices under them. I understand that sending out a read property request requires confirmation, and I believe that means it needs to go to a specified address. (not globally broadcasted). Sending out a whois request I can ignore the responses I get that do not have the device ID I am looking for, but I don't know how to implement something similar with a readproperty requests.  Is their a way to either ignore the wrong device id's or specify which device id the request goes to?

JoelBender, 2016-04-22:
If I understand your topology correctly, there is a BACnet/IP router with an IP address, which is a router to three other BACnet networks.  For the sake of argument, let's say that you and the router are on BACnet network 1 and the three other networks are ARCNET networks 10, 11, and 12.  Each of those other networks have 20-100 devices.  You are correct, Read Property is a confirmed service, so they will be unicast, not broadcast, messages.

The WhoIs service is confirmed and typically broadcast.  In your topology, you should be sending global broadcast messages so they are picked up by the router and re-broadcast on each of the other connected networks.  The IAm messages you get back may be broadcast or unicast, depending on what revision of the standard they used.  Later revisions use unicast messages to help minimize the total volume of broadcast traffic.

You are also correct, you may receive IAm messages from devices that fall outside the device instance low limit and device instance high limit, there are multiple scenarios that can cause a device to broadcast an I-Am.  You should cache the BACnet source addresses of the I-Am responses that you care about and simply discard the rest.  In BACnet terminology you would have then completed the "device address binding" process.

Let's say that you have device 3452 which is on network 10 at ARCNET address 45.  The I-Am response will be presented to your application as 10:45, and the network layer of the application will cache the IP address of your router as the router to network 10.

Subsequent read requests will be unicast messages to 10:45.  When the request makes it all the way down through the stack it will be an Original Unicast at the BVLL layer, and have a DNET/DLEN/DADR of 10/1/45 at the network layer.  The response will also be an Original Unicast from the router, with an SNET/SLEN/SADR of 10/1/45 as you might expect.

JoelBender, 2016-12-01:
I believe this question has been answered.

closed 2016-12-01 by JoelBender