JoelBender/bacpypes #87: Split atomic file services into a separate module

JoelBender, 2016-08-25:
There pieces of the atomic read/write file services that are scattered around in the `app` module and sample applications.  Unify the service side into a `service` module.

See the [Service Subpackage](https://sourceforge.net/p/bacpypes/mailman/message/35015580/) topic in the mailing list.

ChristianTremblay, 2016-09-10:
See issue-87

ChristianTremblay, 2016-09-10:
As this include iocb, API change + services, maybe the issue could be renamed ?

ChristianTremblay, 2016-11-19:
This feature brings really big changes to the API.
Documentation updates have been incorporated into this issue (instead of working in a specific "doc" branch. Those doc updates do not cover only the new `service` API change... But this big change served as a good prétexte to revise everything.

JoelBender, 2016-12-01:
This is finished in #87 moved into `stage` and ready to be released.

closed 2016-12-01 by JoelBender