JoelBender/bacpypes #90: Segmentation handling v14 / Problem with requests

ChristianTremblay, 2016-08-26:
Working request creates this output in Wireshark in the APDU portion :
.100 .... = Max Response Segments accepted: 16 segments (4)
.... 0100 = Size of Maximum ADPU accepted: Up to 1024 octets (4)
The new version 14.1 creates wrong request :
.000 .... = Max Response Segments accepted: Unspecified (0)
.... 0000 = Size of Maximum ADPU accepted: Up to MinimumMessageSize (50 octets)

ChristianTremblay, 2016-08-26:
See  f247ab3 for a start

ChristianTremblay, 2016-08-26:
When maxSegmentsAccepted is not defined, we should start with 1024... not 50 octets

JoelBender, 2016-08-26:
I think you mean when `maxApduLengthAccepted` is not defined, and in that case there should be a default value provided in the `LocalDevice.__init__` function.  I added an additional check to make sure the value is at least 50.

I double checked the encode function and the check was wrong, the encoded value should represent the largest value less than or equal to the property, Clause 12.11.18.

ChristianTremblay, 2016-09-10:
This can be closed ?

closed 2016-11-19 by ChristianTremblay