JoelBender/bacpypes #91: BACNET Server Device

mtroian, 2016-09-01:
I need to develop a Bacnet Server Device with Notification Class Objects. Is there any python program example I can use?
Attached a screenshot of OPCServer where Bacnet device with notitication class object are displayed.

[image]

JoelBender, 2016-09-01:
If you just want to add some objects to your application it's not too hard (untested):

    from bacpypes.primitivedata import Unsigned
    from bacpypes.constructeddata import ArrayOf, SequenceOf
    from bacpypes.basetypes import Destination, Recipient

    some_destination = Destination(
        validDays=[1, 1, 1, 1, 1, 1, 1],
        fromTime=(0, 0, 0, 0),
        toTime=(23, 59, 59, 99),
        recipient=Recipient(device=('objectIdentifier', 12345)),
        processIdentifier=1,
        issueConfirmedNotifications=False,
        transitions=[1, 1, 1],
        )
    some_object = NotificationClassObject(
        objectIdentifier=('notificationClass', 1),
        objectType='notificationClass',
        objectName='Some Notification Class',
        notificationClass=1,
        priority=ArrayOf(Unsigned)([255, 255, 255]),
        ackRequired=[1, 1, 1],
        recipientList=SequenceOf(Destination)([some_destination]),
        )

I'm working on alarm and event services which will enable a BACpypes server to support alarm and event notifications in a separate branch.

JoelBender, 2016-12-01:
ping @mtroian, is this sufficient or do you need something else?

closed 2018-07-04 by JoelBender

shashankgowdasd, 2023-11-28:
Hi,
Thanks @JoelBender for the suggestion, which helped me in implementing the writemultiple property to my server in the last conversation

Currently I required to implement the Alarm and event services to my BACnet server, So can you please guide in regarding that ....

pooyahmt, 2024-01-22:
> Hi, Thanks @JoelBender for the suggestion, which helped me in implementing the writemultiple property to my server in the last conversation
> 
> Currently I required to implement the Alarm and event services to my BACnet server, So can you please guide in regarding that ....

any progress?