JoelBender/bacpypes #93: Unique source folder for Python 3

ChristianTremblay, 2016-09-10:
I propose using a unique folder for Python 3 (py30) instead of restricting to 3.4
Python 3.5 is there.... 3.6 is already dev
Except if we someday use async/wait features, we could add another folder starting at the version with all the features used... But it's not the case actually.

We then will be able to test bacpypes easily toward multiple v3 using Travis for example.

ChristianTremblay, 2016-09-10:
see 50e59d3

JoelBender, 2016-12-01:
I have been symlinking `py35` to `py34` for the newer OS's like Ubuntu 16.04.1, so I don't mind that going forward (I would probably call it `py3x`).  What do you think about having a version to directory map `source_folder = version_to_directory[sys.version_info[:2]]` to make sure that for versions we haven't tested, like 3.3, 3.2, etc., it's a little clearer that they are not supported?

JoelBender, 2017-03-09:
I just stumbled over a workstation that had 2.5 installed and found out that the `setuptools` module I installed uses`with` which isn't supported.  However, it still has `distutils.core` so by uninstalling `setuptools` I was able to install and run some manual tests.  w00t!

Which brings me back around to this issue.  Having a version to directory map would be very simple so `(3, 4)`, `(3, 5)` and `(3, 6)` all map to `py3x`.  That does not solve the problem of creating a version that switches from using its own select loop to asynchronous generators and comprehensions, or a version that runs on [Twisted](https://twistedmatrix.com), [Tornado](http://tornadoweb.org), [gevent](http://www.gevent.org), or any of the others.  Maybe these things should be separate GitHub projects.

JoelBender, 2017-03-09:
Note to self: supporting 2.5 better means not providing `setup_requires`, `zip_safe`, `include_package_data`, `install_requires`, `test_suite`, and `tests_require` keyword parameters to the `setup()` function.  They currently generate warnings.

JoelBender, 2017-03-28:
Note to self: after installing 3.6 on Ubuntu development box, without any additional packages, it's generating the same warning messages as the statement above, and doesn't understand the `test` command.

ChristianTremblay, 2017-03-31:
I would definitly use a py3x folder and accept everything above 3.4 (not below).
Any new version using funky AIO stuff could be in it's own folder... or as you said, a new project maybe... or plugins ? stuff that would use existing modules from the installed version.... but if you select import bacpypes.aio, you will get a async version...

JoelBender, 2017-03-31:
I just pushed this bit of code to the `issue-93` branch (which is waaaay behind otherwise):
```
# different source folders
version_info = sys.version_info[:2]
source_folder = {
    (2, 5): 'py25',
    (2, 6): 'py25',
    (2, 7): 'py27',
    (3, 4): 'py34',
    (3, 5): 'py34',
    }.get(version_info, None)
if not source_folder:
    raise EnvironmentError("unsupported version of Python")
if not os.path.exists(source_folder):
    raise EnvironmentError("broken distirbution, looking for " +
        repr(source_folder) + " in " +
        os.getcwd()
        )
```
The idea is to make version mapping explicit rather than constructed.

JoelBender, 2017-03-31:
I renamed the branch from `issue-93` to `93-source-folder`.  We are using [GitLab](https://about.gitlab.com/) in my office and when you click on the "make a branch for this issue" these are the kinds of branch names it comes up with.  I like it because more than once I've been confused about which branch was for which problem.

JoelBender, 2017-05-01:
I pushed up this change ready for the next release.

closed 2017-05-01 by JoelBender