JoelBender/bacpypes #95: Issues with Date object

sangeeths, 2016-09-27:
Hi Joel, 

I see two issues with [Date](https://github.com/JoelBender/bacpypes/blob/master/py27/bacpypes/primitivedata.py#L1242) implementation.

- When a Date object is created with year=2016, it is not storing the value as 2016-1900=116 (like how it does if we pass a str) - refer [here](https://github.com/JoelBender/bacpypes/blob/master/py27/bacpypes/primitivedata.py#L1247). But it adds 1900 to the currently stored value for year in ```__str__``` - [here](https://github.com/JoelBender/bacpypes/blob/master/py27/bacpypes/primitivedata.py#L1403-L1406)

```
>>> 
>>> from datetime import datetime
>>> print datetime.now().year
2016
>>> 
>>> from bacpypes.primitivedata import Date
>>> d1 = Date(year=datetime.now().year)
>>> 
>>> print d1
Date(3916-*-* *)
>>> 
>>> 
```

- When a str is passed to initialize a Date object, it looks like the calculated ```day_of_week``` is not accurate. 

```
>>> 
>>> from bacpypes.primitivedata import Date
>>> 
>>> from datetime import datetime
>>> 
>>> dt_str = str(datetime.now().date())
>>> print dt_str, datetime.now().date().strftime("%A")
2016-09-27 Tuesday
>>> 
>>> frm_str = Date(dt_str)
>>> print frm_str
Date(2016-9-27 mon)
>>> 
```

Output from console.

```
(bacpypes)[~/code/bacpypes] sangeeth @ space -> cal 9 2016
   September 2016
Su Mo Tu We Th Fr Sa
             1  2  3
 4  5  6  7  8  9 10
11 12 13 14 15 16 17
18 19 20 21 22 23 24
25 26 27 28 29 30

(bacpypes)[~/code/bacpypes] sangeeth @ space -> date
Tue Sep 27 13:01:09 IST 2016
(bacpypes)[~/code/bacpypes] sangeeth @ space -> 
```

Thank you,

Sangeeth

JoelBender, 2016-10-15:
The value of a `Date` object is four element tuple of the values when it is encoded and there is no error checking on the keyword parameter values when you pass them into `__init__()`.  The year is a specific year when the value is 0..254 (1900..2154) or 255 which is "any", used in schedules.  I decided not to check the parameters because there are so many weird values (like the special months and days).  So `Date(year=2016)` is incorrect.

The other piece I can't reproduce.

```
>>> from datetime import datetime
>>> dt_str = str(datetime.now().date())
>>> dt_str
'2016-10-15'
>>> 
>>> frm_str = Date(dt_str)
>>> print frm_str
Date(2016-10-15 sat)
>>> 
>>> print Date('2016-09-27')
Date(2016-9-27 tue)
>>> 
```

I wonder if it's because your in the IST time zone?  What does this look like for you?

```
>>> import time
>>>
>>> today = time.mktime( (2016, 9, 27, 0, 0, 0, 0, 0, -1) )
>>> print today, time.gmtime(today)
1474948800.0 time.struct_time(tm_year=2016, tm_mon=9, tm_mday=27, tm_hour=4, tm_min=0, tm_sec=0, tm_wday=1, tm_yday=271, tm_isdst=0)
>>> 
```

JoelBender, 2018-04-07:
Any updates?  Are there bugs in managing time zones?

closed 2018-07-04 by JoelBender