JoelBender/bacpypes #96: Schedule Object not working as expected

anishparanjpe, 2016-10-14:
Hi, I am creating a bacnet wrapper on top of some bacnet incompatible devices. I would like to control these devices using schedule objects. I  created one of them as follows

```python
class TestScheduleObject(ScheduleObject):

    properties = [
        ReadableProperty(
            'effectivePeriod',
            DateRange,
            default=DateRange(startDate="2016-10-14", endDate="2017-10-14")
        ),
        OptionalProperty(
            'weeklySchedule',
            ArrayOf(DailySchedule),
            default=ArrayOf(DailySchedule)([
                DailySchedule(daySchedule=[TimeValue(time="08:46", value=Real(5))]),
                DailySchedule(daySchedule=[TimeValue(time="08:46", value=Real(6))]),
                DailySchedule(daySchedule=[TimeValue(time="08:46", value=Real(7))]),
                DailySchedule(daySchedule=[TimeValue(time="08:46", value=Real(8))]),
                DailySchedule(daySchedule=[TimeValue(time="08:46", value=Real(9))]),
                DailySchedule(daySchedule=[TimeValue(time="08:46", value=Real(10))]),
                DailySchedule(daySchedule=[TimeValue(time="08:46", value=Real(11))])
            ])
        ),
        ReadableProperty(
            'listOfObjectPropertyReferences',
            SequenceOf(DeviceObjectPropertyReference),
            default=SequenceOf(DeviceObjectPropertyReference)([
                DeviceObjectPropertyReference(
                    objectIdentifier=("analogValue", 0),
                    propertyIdentifier="presentValue",
                    propertyArrayIndex=0,
                    deviceIdentifier=("device", 599)
                )
            ])
        ),
        ReadableProperty(
            'outOfService',
            Boolean,
            False
        )
    ]

    def __init__(self, **kwargs):
        if _debug: TestScheduleObject._debug("__init__ %r", kwargs)
        ScheduleObject.__init__(self, **kwargs)
```

I added the above object as well as the referenced analog-value object to the application; however, the presentValue of the analog-value object is not updated according to the schedule.

Am I missing any parameters to the ScheduleObject, or has the schedule functionality not been implemented as yet?

Thank you.

JoelBender, 2016-10-15:
Like most of the other objects, the ScheduleObject is just a framework for knowing what properties it has and their datatype.  An implementation would be something like the `LocalDeviceObject` that builds on the framework, and that doesn't exist.  If you would like to write one, it would be something like this:

```
    class LocalScheduleObject(ScheduleObject, Task):
        ...
        def __init__(self, **kwargs):
            ...
            self.install_task(0)

        def process_task(self):
            # code goes here
            self.presentValue = current_value
            self.install_task(next_transition)
```

The idea is that the schedule "runs" whenever the value is supposed to change based on the current time, then schedules itself to run again the next time it is supposed to change.  This is non-trivial because of all the special values that `Date` objects can have, in your case you are only using the `weeklySchedule` so it could be easier.

If the `weeklySchedule` is expected to be writable, then you'll need a hook to "run" it by calling `deferred(obj.process_task)` or `obj.install_task(0)` when the property value changes.  That is much easier to write in the upcoming version (0.15.0) than it is now.

dhoomakethu, 2017-11-08:
@JoelBender Do you have a working example of writing to schedules and triggering the schedule at the given time ? I am trying to use `RecurringTask`  but the code fails at initialization with `PropertyError: taskInterval`

```
class BacnetScheduleObject(RecurringTask, ScheduleObject):
    def __init__(self, **kwargs):
        interval = kwargs.get("interval", 3)
        # RecurringTask.__init__(interval)
        # ScheduleObject.__init__(self)

        # install the task
        self.install_task()
```

```
  File "/simulation/schedule.py", line 21, in __init__
    self.install_task()
  File "/lib/python2.7/site-packages/bacpypes/task.py", line 188, in install_task
    if self.taskInterval is None:
  File "/lib/python2.7/site-packages/bacpypes/object.py", line 456, in __getattr__
    prop = self._attr_to_property(attr)
  File "/lib/python2.7/site-packages/bacpypes/object.py", line 443, in _attr_to_property
    raise PropertyError(attr)
PropertyError: taskInterval
```

JoelBender, 2017-11-10:
You are running into an attribute name collision problem, attributes of objects that are a subclass of `Object` are expected to be properties.  I didn't think through my earlier response, so it's misleading and I apologize.  I'm starting a new branch for sample code that will be a better starting point.

dhoomakethu, 2017-11-10:
@JoelBender  Thanks, meanwhile I also started working on alternative to RecurringTask with [schedule](https://github.com/mrhwick/schedule), though I have some issues with it. Its still a working workaround as of now, the GIST is available here. But I will be looking forward for a solution with RecurringTask.

JoelBender, 2017-11-10:
Back a number of years ago I started on [ScheduleObject.py](https://sourceforge.net/p/bacpypes/code/HEAD/tree/trunk/samples/ScheduleObject.py) sample code but didn't do anything with it after that.  The recurring task part isn't so bad (yet) but the `AnyAtomic` beast is tripping up `WriteProperty` and I don't want to break any of the COV notification code.  This is quite the rabbit hole!

dhoomakethu, 2017-11-10:
Yeah, I actually followed the leads from that code , The WriteProperty is working fine for me as of now (though not extensively tested, still under development).

```
class BacnetScheduleObject(ScheduleObject):
    def __init__(self, **kwargs):
        super(BacnetScheduleObject, self).__init__(**kwargs)

    def ReadProperty(self, propid, arrayIndex=None):
        return super(BacnetScheduleObject, self).ReadProperty(propid,
                                                             arrayIndex)

    def WriteProperty(self, propid, value, arrayIndex=None, priority=None, direct=False):
        if propid in ["weeklySchedule", "exceptionSchedule"]:
            if propid == 'weeklySchedule':
                value = ArrayOf(DailySchedule)(value)
            elif propid == "exceptionSchedule":
                value = ArrayOf(SpecialEvent)(value)
        return super(BacnetScheduleObject, self).WriteProperty(propid, value, arrayIndex, priority, direct)

    @classmethod
    def make_mutable(cls):
        for prop in cls.properties[:]:
            if isinstance(prop, OptionalProperty):
                prop.mutable = True
            elif prop.identifier == "priorityForWriting":
                prop.mutable = True
```

and in my application I create the bacnet object like this

```
....

bacnet_obj = BacnetScheduleObject
            # DailyScheduleArray = ArrayOf(DailySchedule)
            weeklySchedule = [DailySchedule(daySchedule=[])]*7
            exceptionSchedule = []
            presentValue = AnyAtomic(
                Unsigned(random.randint(*unsigned_range)))
            listOfObjectPropertyReferences = SequenceOf(DeviceObjectPropertyReference)([
                DeviceObjectPropertyReference(
                    objectIdentifier=("analogValue", 1),
                    propertyIdentifier="presentValue",
                    propertyArrayIndex=0,
                    deviceIdentifier=("device", address)
                )
            ])
            priorityForWriting = Unsigned(9)

this_app.add_object(bacnet_obj(
                objectIdentifier=point.instance,
                objectName=point_name,
                presentValue=presentValue,
                weeklySchedule=weeklySchedule,
                listOfObjectPropertyReferences=listOfObjectPropertyReferences,
                exceptionSchedule=exceptionSchedule,
                priorityForWriting=priorityForWriting

            ))
```

JoelBender, 2017-11-10:
The part about nesting the value inside any AnyAtomic instance like `presentValue = AnyAtomic(Unsigned(1))` shouldn't have to happen, `presentValue = Unsigned(1)` should be enough.  The `priorityForWriting` is already unsigned, so you shouldn't wrap that in an `Unsigned` instance.

dhoomakethu, 2017-11-10:
Cool, I will try those changes

JoelBender, 2017-11-11:
In [this commit](https://github.com/JoelBender/bacpypes/commit/30c137cadad2b82814c900b19469e55f21684eaa) I made `AnyAtomic` a subclass of `Atomic` so the `sandbox/sample_schedule_object.py` works as I want it to.  The `ReadWriteProperty.py` sample application has an additional update so the "value" can be lots of atomic types.

There isn't any type checking, so the `presentValue` can be `Integer(12)` and the `scheduleDefault` can be `Real(13.5)`.

JoelBender, 2017-12-07:
This has been merged into #144, to be continued.

closed 2017-12-07 by JoelBender