JoelBender/bacpypes #97: Uncaptured python exception when network disconnected

PeteDavidson, 2016-10-26:
When I disconnect my network cable I get the following:

error: uncaptured python exception, closing channel <bacpypes.udp.UDPDirector 157.184.139.141:47808 at 0x7fee434b7d50> (<class 'socket.error'>:[Errno 22] Invalid argument [/usr/lib/python2.7/asyncore.py|write|91] [/usr/lib/python2.7/asyncore.py|handle_write_event|468] [/usr/local/lib/python2.7/dist-packages/bacpypes-0.14.1-py2.7.egg/bacpypes/udp.py|handle_write|218])

Maybe need to catch socket.error? Catching OSError does not handle it.

JoelBender, 2016-10-26:
Acknowledged - I'll add this to the 0.15 release which is almost ready to roll.

JoelBender, 2016-10-26:
In Python 3.2 they made socket.error a subclass of IOError, so I think that when I was working on Python 3.4 testing and came across it I "back ported" (a fancy word for copy/paste) the exception handler into the other versions.

JoelBender, 2016-12-01:
This is in `stage` (as a "release candidate"), I should have mention the issue number when I committed the fix.

closed 2016-12-01 by JoelBender