JoelBender/bacpypes #98: How to use bacpypes with BACnet ethernet

hoangdungt2, 2016-10-27:
Hi,

I'm new to python and bacpypes, I tried to look at the example but seems to me all are for BACnet/IP. I wonder if any of you can help to write a BACnet ether example, I would be very grateful.

Thank you very much

JoelBender, 2016-10-27:
There are two ways that I have found for doing this work, raw sockets on Linux and winpcap on Windows (I prowled around the source code for the current version of [VTS](https://sourceforge.net/projects/vts/) to see how I had accomplished this before and I can't find it yet).  If you would like to try a raw sockets implementation I could include in the project that would be awesome!

Start by forking the project, duplicate the `udp.py` module as `eth.py`, rename *UDP* to *Ethernet* (like `UDPActor` to `EthernetActor`), change the socket type request in the director:

```
        # ask the dispatcher for a socket
        self.create_socket(socket.AF_PACKET, socket.SOCK_RAW)
```
Add to the `handle_read()` function after receiving the packet to split out the destination address, source address, length, and DSAP and SSAP fields.  Check to make sure that DSAP and SSAP are 0x82 for BACnet.  The rest of the content goes into the pduData field for a PDU going up the stack.

Add to the `handle_write()` function after getting the request to build an Ethernet frame with the same header fields (make sure you get the length right) and the rest of the packet is the pduData attribute.

In the BVLL service layer there is code for recognizing Annex-H vs Annex-J UDP packets, which you won't need, but there is code for recognizing when the source and destination addresses are local broadcast, so you'll need to translate those as well.

I'll leave this issue open for your comments and for me to add this in the future, otherwise you can close it and let me know what your fork is called.

JoelBender, 2016-10-28:
The [BACnet Firewall Router](https://sourceforge.net/p/bfr) project has an implementation in [BFREthernet.cpp](https://sourceforge.net/p/bfr/code/HEAD/tree/trunk/BFR/BFREthernet.cpp) that should be very close to what is needed.  I *knew* I had that code someplace :-)

hoangdungt2, 2016-11-03:
Hi JoelBender,

Thank you very much for the help. 
To be honest, I'm not very familiar with BACnet raw data management.
Actually, I was hoping bacpypes can allow me to just specify the ethernet adapter then I can just do whois then read/write property directly to the discovered devices.
I see this https://sourceforge.net/projects/yetanotherbacnetexplorer/ allow me to but it is in C# so I was hoping to have something similar but in Python.
Thanks a lot.

JoelBender, 2016-11-03:
From [BacnetTransportEthernet.cs](https://sourceforge.net/p/yetanotherbacnetexplorer/code/HEAD/tree/trunk/Yabe/BacnetTransportEthernet.cs) it is stacked on top of [SharpPcap](https://github.com/chmorgan/sharppcap) which is in turn stacked on WinPcap, and there is a [winpcapy](https://github.com/orweis/winpcapy) project that is very similar, so there is definitely a way forward.