JoelBender/bacpypes #99: SubscribeCOVPropertyRequest sequence Elements have incorrect optional flag

karolgil, 2016-11-16:
BACnet standard (EN ISO 16484-5) states that request SubscribeCOVPropertyRequest sent without issueConfirmedNotifications and/or lifetime and/or covIncrement is a valid request (e.g. without issueConfirmedNotifications  and lifetime it means subscription cancellation).
I think it would be wise to change this request class to:

`class SubscribeCOVPropertyRequest(ConfirmedRequestSequence):`
`    serviceChoice = 28`
`sequenceElements = \`
        `[ Element('subscriberProcessIdentifier', Unsigned, 0)`
     `   , Element('monitoredObjectIdentifier', ObjectIdentifier, 1)`
  `      , Element('issueConfirmedNotifications', Boolean, 2, True)`
   `     , Element('lifetime', Unsigned, 3, True)`
`        , Element('monitoredPropertyIdentifier', PropertyReference, 4)`
 `       , Element('covIncrement', Real, 5, True)`
 `       ]`

to be compliant with the standard.

ChristianTremblay, 2016-11-17:
Can you make a PR ? Joel will evaluate it and merge if accepted.

karolgil, 2016-11-17:
I have a question regarding that - I'm kind of new to github and I don't know the whole process yet.
I've pulled this repo, made the changes locally on the local branch "issue-99" - but I'm not able to push that branch to origin. How can I get the permission to do that?

Sorry for the trouble...

JoelBender, 2016-11-17:
Karol,

> I have a question regarding that - I'm kind of new to github and I don't know the whole process yet.

No worries, everybody start the bottom of the learning curve.  And soon as you know that one, there is another!

> I've pulled this repo, made the changes locally on the local branch "issue-99" - but I'm not able to push that branch to origin. How can I get the permission to do that?

If I add you as a "collaborator" then you should be able to, give you write access to the repository.  What Christian is talking about means "forking" the project into your own namespace, making all the necessary changes, then when you think it ready there is a "generate a pull request" on the web page.

Let's try the pull request thing.

    https://help.github.com/articles/creating-a-pull-request-from-a-fork/

Thank you for helping!

Joel

JoelBender, 2016-11-18:
I received your pull request and merged it in.  Thank you!

closed 2016-11-18 by JoelBender