JoelBender/bacpypes #100: whois

erdongwei, 2016-11-18:
Hi, I'm new to BACpypes, and have some problems about whois services. When use whois service, the devices response to whois send broadcast message, not unicast message to the pc that requests whois service, and the pc cannot get these broadcast messages.  Any ideas? Thanks.

JoelBender, 2016-11-18:
You are correct, the WhoIsRequest is a broadcast service, and the "old school" response (which is an IAmRequest) was also sent as a broadcast.  More modern implimentations send the response back as a unicast message and you application should expect both types of behavior.  This was done to keep broadcast traffic manageable with large networks.

Look carefully at the address in the INI file and notice that it is in CIDR notation, like `1.2.3.4/24`.  The implicit broadcast address is `1.2.3.255` so to receive these broadcast messages the application will open *two* sockets, one for sending all messages and for receiving unicast ones, the other just for receiving broadcasts.  The `UDPMultiplexer` instance ties these two sockets together.

In the samples directory there is a `samples\UDPConsole.py` application. Run that program on two different machines with the correct IP address, or localhost `127.0.0.1`, or `any`.

erdongwei, 2016-11-18:
Hi, thanks for the quick response. I can run UDPConsole.py on machine B with correct IP to enable machine A to receive broadcast message, right? And what if I only have one machine to use, is there a possible way that I can enable itself to receive broadcast message during whoisRequest? Maybe combine UDPConsole.py and WhoIsIAm.py together?

JoelBender, 2016-11-18:
The stumbling block is UDP and that two applications running on the same machine can't have the same socket open.  One machine will have port 47808 and the other will have 47809, and it's easy to send unicast packets, but they are in a different broadcast domain.

Are you familiar with BACnet BBMD's?  You could have one application (A) running as a BBMD (BACnet Broadcast Management Device) and then have the other one (B) register as a foreign device.  Then all of the broadcast messages B wants to send are unicast to the BBMD as `distributeBroadcastToNetwork` messages and the BBMD will rebroadcast it.

Another option is to run some virtualization software (I use [VirtualBox](https://www.virtualbox.org/)) and run the applications on two different virtual machines.  And another option is to build a virtual network from the components in the `vlan` module and combine the applications into one program.

erdongwei, 2016-11-18:
Sorry, I don't think I have made my question clear. My question is how to make the machine requesting WhoIs service to receive broadcast messages from other devices, thus, get the information of the devices of the internetwork.

JoelBender, 2016-11-18:
Just make sure the address you pass to the application is in CIDR notation, like `192.168.1.4/24` and the stack will listen on both the unicast address and the broadcast address.  The unicast socket will be bound to `('192.168.1.4', 47808)` and the broadcast socket will be bound to `('192.168.1.255', 47808)`.

erdongwei, 2016-11-18:
Yes, I pass the CIDR notation, like 1.2.3.4, and I can see from wireshark that some devices response to 1.2.3.255, but the whoIsIAm.py application didn't get that response.

erdongwei, 2016-11-18:
Yes, the address I pass is in CIDR notation, like 1.2.3.4/24, and I can see from the wireshark that some devices send message to 1.2.3.255, but the WhoIsIAm application doesn't get that response.

erdongwei, 2016-11-18:
I found the reason why it cannot receive the broadcast message, it's because of the firewall is enabled, after I disabled it, everything works fine. Thanks.

ChristianTremblay, 2016-11-19:
I will add a warning notice about firewall in the doc

ChristianTremblay, 2016-11-19:
Will be part of the doc of 0.15

closed 2016-11-19 by ChristianTremblay

erdongwei, 2016-11-22:
Okay, thanks for the help.