JoelBender/bacpypes #102: Calculated Services Supported

JoelBender, 2016-12-01:
Most of the sample applications, and real ones, have code like this:
```
    # get the services supported
    services_supported = this_application.get_services_supported()
    if _debug: _log.debug("    - services_supported: %r", services_supported)

    # let the device object know
    this_device.protocolServicesSupported = services_supported.value
```
It seems like this should be included at the end of the `Application.__init__()` function and it would save boilerplate code.  There would be a little extra work to do when services are added dynamically, and for those applications that want to present services supported as something custom, they would add it themselves to the device object and the automatic calculation would leave it alone.

JoelBender, 2017-12-18:
After working on the `CurrentPropertyList` property list functionality that returns the value dynamically, that would be a good model to follow for this issue.  Just get the services supported at the time the property is requested from the `LocalDeviceObject`.

amrx101, 2018-01-18:
@JoelBender , I can take it up if its up for grabs.

JoelBender, 2018-01-19:
Go for it! :+1:

JoelBender, 2018-04-07:
There is now a `CurrentProtocolServicesSupported` class in the `bacpypes.device.local` module in 0.17.0 that is used for the `LocalDeviceObject` and it simplifies lots of sample applications.  Enjoy!

closed 2018-04-07 by JoelBender