JoelBender/bacpypes #103: Encode Response Error

charador, 2017-01-19:
Hello!

I'm trying to write a function response() to encode response datapayload and send it out. However, I got an error in the function  `npdu.encode(obnpdu)`
It says that in `npdu.py`:
`if self.npduDADR.addrType == Address.remoteStationAddr:
AttributeError: 'tuple' object has no attribute 'addrType'`
I'm not sure what's the correct way to encode the response payload. Thanks!

Here is my code:
`    def response(self, response_apdu, address):
        if response_apdu is None:
            return
        try:
            apdu = APDU()
            response_apdu.encode(apdu)
        except Exception, err:
            logger.error('EncodeError: apdu: %s', err)
            return apdu
        #
        #
        apdu_type = apdu_types.get(apdu.apduType)
        logger.info("apdu_type is %s", apdu_type)

        # "lift" source and destination address
        npdu = NPDU()

        if apdu.pduSource:
            logger.info('setup npdu.npduSADR')
            npdu.npduSADR = apdu.pduSource
        if apdu.pduDestination:
            logger.info('setup npdu.npduDADR')
            npdu.npduDADR = apdu.pduDestination

        try:
            apdu.encode(npdu)
        except Exception, err:
            logger.error('EncodeError: npdu: %s', err)

        npdu.npduHopCount = int(255)#should it set to something else?
        obnpdu = OriginalUnicastNPDU()
        if (apdu.apduType == ConfirmedRequestPDU.pduType or apdu.apduType == ComplexAckPDU.pduType or apdu.apduType == SimpleAckPDU.pduType) :
            logger.info('Isinstacne Unicast.')
            obnpdu = OriginalUnicastNPDU()
        elif (apdu.apduType == UnconfirmedRequestPDU.pduType):
            logger.info('Isinstacne Broadcast.')
            obnpdu = OriginalBroadcastNPDU()
        elif (apdu.apduType == DistributeBroadcastToNetwork.bvlciType):
            logger.info('Isinstacne DistributeBroadcastToNetwork.')
            obnpdu = DistributeBroadcastToNetwork()
        elif (apdu.apduType == ForwardedNPDU.bvlciType):
            logger.info('Isinstacne ForwardedNPDU.')
            npdu.pduSource = apdu.bvlciAddress
        else:
            logger.info('Error: blvi function not implemented.')

        try:#error here! 'npdu' has no attribute 'addrType'
            npdu.encode(obnpdu)
        except Exception, err:
            logger.error('Obnpdu encode error: %s', err)

        bvlp = BVLPDU()
        try:
            obnpdu.encode(bvlp)
        except Exception, err:
            logger.error('bvlp encode error: %s', err)

        pdu = PDU()

        try:
            bvlp.encode(pdu)
        except Exception, err:
            logger.error("pdu encode error: %s", err)

        if isinstance(response_apdu, RejectPDU) or isinstance(response_apdu, ErrorPDU):
            self.datagram_server.sendto(pdu.pduData, address)
        else:
            if pdu.pduDestination == '*:*':
                # broadcast
                # sendto operates under lock
                self.datagram_server.sendto(pdu.pduData, ('x.x.x.255', address[1]))
            else:
                # sendto operates under lock
                self.datagram_server.sendto(pdu.pduData, address)
            logger.info('Bacnet response sent to %s (%s:%s)',
                        response_apdu.pduDestination, apdu_type.__name__, self._response_service)`

charador, 2017-01-19:
code format updated

JoelBender, 2017-01-20:
The `npdu.npduDADR` contents should be something of type `Address` and the error message says it's a tuple, so rewinding back through the code you set the value to `apdu.pduDestination` so that is also a tuple.  Note also that the `npduDADR` field is only used when the destination is a remote station (specific device on a different network), remote broadcast (all the devices on a different network) or a global broadcast.  So not just any `Address` instance is acceptable.

The fact that you set the `apdu.pduDestination` to a tuple makes me speculate that when the apdu is build the destination is given a socket tuple like ('1.2.3.4', 5), so you could maybe you should build something like `Address(('1.2.3.4', 5))` which converts the tuple to an address object.  To make a remote address you have to do it in two steps:

```
>>> x = Address(('1.2.3.4', 5))
>>> y = RemoteAddress(6, x.addrAddr)
```

And to the other comment question; yes, the hop count always starts out at the maximum value, unless you are doing something special with network paths and routing and you want the "hops" to be clipped short.

charador, 2017-01-22:
Thanks Joel!

1. about the address: In your example, 

>  x = Address(('1.2.3.4', 5))
>  y = RemoteAddress(6, x.addrAddr)

I guess both `5` and `6` refer to the port number(47808)? or they are `device instance numbers`?

Is there a way to get the Remote/Global Broadcast Address from Source Address?
Right now I have to manually specify the remote broadcast address as:
`self.datagram_server.sendto(pdu.pduData, ('x.x.x.255', address[1]))`

2. I try to implement different services such as `ReadProperty`, `WriteProperty`, and `Whois` with the `response` function in the first post. `response_apdu` in `response()` maps to different services. 
I'm able to send IAM service with `response()`; however, the `ReadPropertyACK()` sends as a `ConfirmedReq` rather than `ComplexACK`. 

```
logger.info('response_apdu is %r', response_apdu)
logger.info("apdu.apduType is %s, pduType is %s", apdu.apduType, apdu_type.pduType)
logger.info("apduInvokeID is %s",apdu.apduInvokeID)
```

Shows 
```
response_apdu is <bacpypes.apdu.ReadPropertyACK(12,1) instance at 0x7fc8ed692550>
apdu.apduType == 0x3 (ComplexACK),  apdu_type.pduType == 0x3 (ComplexACK), and InvokeID == 1
```

but when I print out: 
`        logger.info('apdu.pduData is %r', apdu.pduData)`

It shows that I send out `ReadPropertyRequest` as `ComfirmedRequest` rather than `ComplexACK`.
And `InvokeID` resets to `0`.
I spent the whole day trying to fix it.. but I couldn't figure out the reason. 
I guess I may miss something in the `response()` so that it overwrites the header in `response_apdu`.
Can you please help me check if I miss any part in `response()` to encode different types of services: Confirmed/uncofirmed, Request/SimpleACK/ComplexACK. Appreciate it!

JoelBender, 2017-01-23:
1(a) No, the 5 is the port number for UDP traffic so you will see tuples more like `('192.168.0.12', 47808)`.  Device instance numbers are usually larger values, in the case of a nearby large institutional network, they are numbers like 3406712, where 3406 is a code for the building, 34067 is the BACnet network number, and 3406712 is a device on that network.  It's following a general plan of YYYYXNN, _but that's a local convention and every BACnet network is different_.

1(b).  If you are using the UDP layer in BACpypes rather than writing directly to sockets, the destination address should be an instance of `LocalBroadcastAddress` and the UDP layer will translate that into the appropriate IP address based on the binding (in your case using `x.x.x.255` for a `x.x.x.0/24` network).

2. There are three general layers of abstraction in the BACpypes library, the *LAN* layer like UDP and virtual LANs, the "network" layer that handles the BACnet networking layer of BACnet routers and routing traffic, and the *application* layer that sends and receives the confirmed and unconfirmed requests, and simple and complex acknowledgements.  The application layer has an additional sub-layer that deals with segmented messages, chopping up big requests into pieces and reassembling them.

I'm not sure I understand what layer you are expecting your application "sit".  Are you writing a client application (that will be sending requests to other BACnet devices for data), a server application (that will be presenting itself on the network as a BACnet device), or both?  You may be attempting to write an application that implements some portion of an entire stack, where your program is going to be responsible for all of the intermediate encoding and decoding.

The layers of abstraction are intended to shield the application layer developer from the network layer, just like the socket library is designed to shield protocol implementations from the IP network layer. Similarly, the application layer has three layers of classes, each one more specific than the others.  A `ReadPropertyRequest` is a kind of `ConfirmedRequestPDU` which is in turn a kind of `APDU`, so depending on where/when you debug the contents of the packet, you'll be seeing a more abstract or more concrete version of the packet.

Because things like "source" and "destination" depend on the layer and context, it's hard to answer your other questions.

ChristianTremblay, 2017-01-25:
In BAC0, I've used a twist to find the subnet mask and calculate the broadcast address automatically : 
https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/functions/GetIPAddr.py

charador, 2017-02-01:
@ChristianTremblay That's a nice way to find broadcast address without using `255.255.255.255`

@JoelBender I've figured out the problem. I'm writing a BACnet/IP server which can decode incoming bacnet requests and send proper response packets. :)

closed 2017-02-01 by charador