JoelBender/bacpypes #105: Who has request limits seems not properly handled.

Shai-tan, 2017-02-27:
Who has request limits seems not properly handled.
The protocol definition describes this parameter as optional. But the following error occurred when my bacpypes server tried to decode APDU.

`2017-02-27 11:00:08,988 - DEBUG - appservice - indication -     - decoding reject: InvalidTag('deviceInstanceRangeLowLimit expected context tag 0',)`

I uses bacpypes 0.15.0 as server and VTS 3.6.5.0 as client.
The issue is resolved by setting optional flag to lowLimit and highLimit in WhoHasLimit object

JoelBender, 2017-02-28:
I agree, the decoding is incorrect.  The `deviceInstanceRangeLowLimit` and `deviceInstanceRangeHighLimit` are required in the sense that in order to decode the `WhoHasLimits` sequence they both need to be provided.  The `limit` element is optional, so when decoding the `WhoHasRequest` it should accept both of them missing and then continue to require the `objectIdentifier` or the `objectName` of the `object`.

There is another case where something similar is happening, the `ReadRangeRequest` has a `range` element which is an optional choice, so if the decoding of the sub-elements fails (like when none of the choices has been specified) the whole thing should still pass.

I think both of these cases can be solved by catching the `InvalidTag` raised during the decode in the section of `constructeddata.py` in `Sequence.decode()` where there is the comment `# some kind of structure` which restores the tag list from its backup and continues on.

JoelBender, 2017-02-28:
The `105-who-has-request-limits` branch has this change, but needs unit tests.

Shai-tan, 2017-02-28:
I just test the branch and I confirm it correct my issue. The only change is that apdu.limit is now NoneType.

During check of this update, I noted that the limit parameters are not check in WhoHasIHaveServices class implementation. Then the application always reply to WhoHas request.

JoelBender, 2017-03-24:
Unit tests will have to wait, there is an application layer test framework I'm working on that still needs help and I don't want to hold up the next release.

closed 2017-03-24 by JoelBender