JoelBender/bacpypes #107: Unable to process warm restart properly.

Shai-tan, 2017-02-28:
Unable to process warm restart properly.
I'm not able to process warmstart without stopping the process instance that contains the server.
The limitation is due to a business part associated to Bacnet server that cannot be stopped when a warmstart is scheduled.

   result = method(*self.path_args, **self.path_kwargs)
  File "/home/calligp/workspace/P5.4/HagerEnergyServerMiddleEnd/src/Servers/HttpInterfaces/Preferences/BacnetServerHandler.py", line 32, in delete
    wResult, wMessage = self.application.GetBacnetAdministrator().RestartServer()
  File "/home/calligp/workspace/P5.4/HagerEnergyServerMiddleEnd/src/Servers/BacnetAdministrator.py", line 675, in RestartServer
    self._ApplicationNotif(HesProcessInterface._IPCTYPE['supervision_reboot'], self.GetAllSettings())
  File "/home/calligp/workspace/P5.4/HagerEnergyServerMiddleEnd/src/Servers/BacnetInterfaces/BacnetApplication.py", line 211, in ProcessAdminRequest
    self.RestartServer(aData)
  File "/home/calligp/workspace/P5.4/HagerEnergyServerMiddleEnd/src/Servers/BacnetInterfaces/BacnetApplication.py", line 94, in RestartServer
    self._Server = BacnetServer(self._Configuration, aData['communication'], self._MiddleEndCore)
  File "/home/calligp/workspace/P5.4/HagerEnergyServerMiddleEnd/src/Servers/BacnetInterfaces/BacnetServer.py", line 103, in __init__
    BIPSimpleApplication.__init__(self, self._LocalDevice, self._Ip)
  File "/home/calligp/workspace/P5.4/POC/bacpypes/py27/bacpypes/app.py", line 824, in __init__
    self.mux = UDPMultiplexer(self.localAddress)
  File "/home/calligp/workspace/P5.4/POC/bacpypes/py27/bacpypes/bvllservice.py", line 91, in __init__
    self.directPort = UDPDirector(self.addrTuple)
  File "/home/calligp/workspace/P5.4/POC/bacpypes/py27/bacpypes/udp.py", line 147, in __init__
    self.bind(address)
  File "/usr/lib/python2.7/asyncore.py", line 342, in bind
    return self.socket.bind(addr)
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 98] Address already in use

In my current implementation i use BIPSimpleApplication, which instantiates UDPMultiplex, which instantiates 2 UDPDirector.
This issue is linked to the "directPort" attribute from UDPDirector.

2 solutions can be considered :
- Set reuse parameter in directPort instantiation.
  A reuse Boolean parameter is available in UDPDirector interface, when reuse flag is set the warmstart is processed properly.
  But in my opinion this change is not safe because it allows two applications to use the same port. That can induce business treatment issue.
- Add interface to unbind properly the socket.
  This solution provides an interface in BIPSimpleApplication, UDPMultiplexer and UDPDirector to force explicit close of the socket.

  I recommend this solution.
  See attached patch [SocketUnbind.txt](https://github.com/JoelBender/bacpypes/files/806273/SocketUnbind.txt)

Environnement : bacpypes 0.14.1 (and 0.15.0) on Linux.

JoelBender, 2017-03-04:
According to the standard, a WARMSTART is an instruction for a device to "reset itself to some predefined initial state."  I would interpret this in the context of a Linux server environment to mean one of two things; the process providing the service exits and a new instance is started, or the process has the ability to dump its existing running environment/configuration and load a new one.  Most services do the former, there are some like [nginx](http://nginx.org/en/docs/control.html) that accept HUP signals to reload without closing existing sockets, allowing workers to continue satisfying existing requests.

I understand why you do not want to use the reuse flag and explicitly close the sockets.  The only change I would like to make is to rename the function `close_socket()` to match PEP8 conventions.  Should the UDPMultiplexer also pass along the close socket to the broadcast port as well?  Does it also suffer from the same reuse issues that the direct port has?

JoelBender, 2017-03-04:
Let me know if the `107-unable-to-process-warm-restart` branch is acceptable and I'll merge this into `stage` for the next release.

Shai-tan, 2017-03-07:
I confirm the branch 107 correct the issue.
To answer to your questions, it is not an issue to keep reuse flag on broadcast port. In fact, if someone wants to run many server instances on the same computer, the reuse flag is required because broadcast port is same for all instance. Only the direct port will be different.

JoelBender, 2017-03-07:
Thank you, merged into `stage`.

commit 363f6124 (2017-03-07): merging issue branch, closes #107
closed 2017-03-07 by JoelBender