JoelBender/bacpypes #109: Sample script that showcases use of bacpypes.app.BIPForeignApplication

nilstoedtmann, 2017-03-09:
e.g. 

Phase 1: Discover existence of BACnet/IP devices (instance-id and IP address) with WHOIS broadcasts via a BBMD.

Phase 2: For each BACnet/IP device found, query the its label (objectName of device object) via unicast read_prop.

JoelBender, 2017-03-09:
There is a new `109-device-discovery` branch with a sample application `DeviceDiscovery.py` applications in it, I'm going to clone this and make it "foreign" to hunt down the `deviceInfoCache` bug.

nilstoedtmann, 2017-03-09:
Adding `self.smap.deviceInfoCache = self.deviceInfoCache` here https://github.com/JoelBender/bacpypes/blob/master/py27/bacpypes/app.py#L526 solves it for us

nilstoedtmann, 2017-03-09:
(We can send a pull request, but it's gonna take a while)

JoelBender, 2017-03-09:
I committed the other sample along with the fix in the `bacpypes.app` module, let me know if it works.

nilstoedtmann, 2017-03-09:
That should do it. We went ahead with a modified clone of `bacpypes.app.BIPForeignApplication` that has exactly your change, and that worked.

JoelBender, 2017-03-24:
Merged into stage

commit 8db3db25 (2017-03-09): propagate the device information cache in the `BIPForeignApplication` like it does in the `BIPSimpleApplication`, should fix #109
closed 2017-03-24 by JoelBender