JoelBender/bacpypes #112: Don't nick SIGTERM

nilstoedtmann, 2017-03-16:
[bacpypes.core](https://github.com/JoelBender/bacpypes/blob/master/py27/bacpypes/core.py#L167) sets bacpypes.core.stop as signal handler for SIGTERM. That stops the application importing bacpypes from catching SIGTERM for a graceful shutdown.

We are happy to call bacpypes.core.stop() in *our* SIGTERM handler.

Or maybe make it configurable.

JoelBender, 2017-03-20:
Humm... how about `sigterm` and `sigusr1` as new keyword parameters to the run function with the existing functions as defaults?
```
def run(spin=SPIN, sigterm=stop, sigusr1=print_stack):
    ...
    if (sigterm is not None) and hasattr(signal, 'SIGTERM'):
        signal.signal(signal.SIGTERM, sigterm)
    if (sigusr1 is not None) and hasattr(signal, 'SIGUSR1'):
        signal.signal(SIGUSR1, sigusr1)
    ...
```
Your applications could pass in `None`.  I'm not sure how many applications rely on the current behavior, apart from mine.  I would like to hear from more users.

nilstoedtmann, 2017-03-21:
:+1: That should work for us. Good not to set the handler at import time.

JoelBender, 2017-03-21:
Ok, this is now sitting in the [112-dont-nick-sigterm](https://github.com/JoelBender/bacpypes/tree/112-dont-nick-sigterm) branch.

nilstoedtmann, 2017-03-21:
BTW you could also check whether a signal has already a handler assigned and skip if so. See e.g. http://stackoverflow.com/a/12018294 . 
I don't have a preference, just saying.

nilstoedtmann, 2017-03-21:
Either way I hope this will find its way into the next release :)

guangyuzhang, 2017-03-23:
Hi Joel,
I want to set the timeout value or stop the core for WhoIs request in WhoIsIAmApplication, so the request can return device list in shorter time. How can I do it ? I think that most of responses will return in less than 1 second, how can I set the timeout value to 5 seconds and then stop the core? Thanks.

JoelBender, 2017-03-26:
I missed this with 0.15.1 out of git branch sync confusion with the different platforms I'm using for testing, but it will be in 0.15.2 shortly.

closed 2017-03-26 by JoelBender

JoelBender, 2017-03-30:
This has introduced a bug, see #119 and continue the discussion...