JoelBender/bacpypes #115: Who-Is Request Timeout

JoelBender, 2017-03-23:
_Redirected to a new issue from a comment by @guangyuzhang_

I want to set the timeout value or stop the core for WhoIs request in WhoIsIAmApplication, so the request can return device list in shorter time. How can I do it ? I think that most of responses will return in less than 1 second, how can I set the timeout value to 5 seconds and then stop the core?

JoelBender, 2017-03-24:
The device address binding service pair of Who-Is and I-Am messages are not confirmed services, so there's nothing like a request/response context that is managed for you [1].  If you want to stop the core at some time in the future, schedule a task to call the `stop()` function, and there's is a handy class in the `bacpypes.task` module for that:
```
from bacpypes.core import run, stop
from bacpypes.task import TaskManager, FunctionTask

# create a task manager so tasks can be scheduled
task_manager = TaskManager()

# create and schedule a call to stop()
stop_task = FunctionTask(stop)
stop_task.install_task(delta=5.0)

print("running...")

# let it run
run()

print("...done")
```
[1] there _could_ be an IOCB context created for the Who-Is that then holds the I-Am responses until a specific timeout occurs, and there is a `###` comment in the `WhoIsIAmServices.who_is()` function where that might happen.

guangyuzhang, 2017-03-24:
It works! Thanks @JoelBender

closed 2017-03-24 by JoelBender