JoelBender/bacpypes #119: sigterm will fail if core.run() is called in a thread

ChristianTremblay, 2017-03-29:
In BAC0, I call the core run function in it's thread. 
It leads to : 
```
Exception in thread Thread-6:
Traceback (most recent call last):
  File "Folder\Anaconda3\lib\threading.py", line 914, in _bootstrap_inner
    self.run()
  File "Folder\Anaconda3\lib\threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "Folder\Anaconda3\lib\site-packages\bacpypes\core.py", line 97, in run
    signal.signal(signal.SIGTERM, sigterm)
  File "Folder\Anaconda3\lib\signal.py", line 47, in signal
    handler = _signal.signal(_enum_to_int(signalnum), _enum_to_int(handler))
ValueError: signal only works in main thread
```
Setting defaults arguments to None will prevent the usage of sigterm until it's explicitly called.

JoelBender, 2017-03-30:
The code for #110 could check to see if it's running in the main thread by checking `threading.current_thread() == threading.main_thread()` in Python3.4 and up or `isinstance(threading.current_thread(), threading._MainThread)` (according to [stackoverflow](http://stackoverflow.com/questions/23206787/check-if-current-thread-is-main-thread-in-python)) but this seems *really* ugly.

I'm tempted to just pull that functionality out of `run()` and put it in new functions called something like `sigterm_stop()`.

JoelBender, 2017-03-30:
Then again, maybe it's not so bad:
```
    # install the signal handlers if they have been provided (issue #112)
    # and if this is running in the main thread (issue #119)
    if isinstance(threading.current_thread(), threading._MainThread):
        if (sigterm is not None) and hasattr(signal, 'SIGTERM'):
            signal.signal(signal.SIGTERM, sigterm)
        if (sigusr1 is not None) and hasattr(signal, 'SIGUSR1'):
            signal.signal(signal.SIGUSR1, sigusr1)
```
This at least keeps the signals installed by default before I broke it.  There is a very minor function name difference with the Python2.5 version, `threading.currentThread()`.

ChristianTremblay, 2017-03-30:
Waht would you think of something like : 
```
@bacpypes_debugging
def run(spin=SPIN, sigterm=stop, sigusr1=print_stack, in_main_thread=True):
    if _debug: run._debug("run spin=%r sigterm=%r, sigusr1=%r", spin, sigterm, sigusr1)
    global running, taskManager, deferredFns, sleeptime

    try:
        # install the signal handlers if they have been provided (issue #112)
        if in_main_thread:
            if (sigterm is not None) and hasattr(signal, 'SIGTERM'):
                signal.signal(signal.SIGTERM, sigterm)
            if (sigusr1 is not None) and hasattr(signal, 'SIGUSR1'):
                signal.signal(signal.SIGUSR1, sigusr1)
    except ValueError:
        raise Exception('Attemp to use Run in another thread. Use argument "in_main_thread=False" to correct')
```

JoelBender, 2017-03-31:
I would rather not add any more parameters, and maybe a warning is more appropriate:
```
    if isinstance(threading.current_thread(), threading._MainThread):
        if (sigterm is not None) and hasattr(signal, 'SIGTERM'):
            signal.signal(signal.SIGTERM, sigterm)
        if (sigusr1 is not None) and hasattr(signal, 'SIGUSR1'):
            signal.signal(signal.SIGUSR1, sigusr1)
    elif sigterm or sigusr1:
        warnings.warn("no signal handlers for child threads")
```
Is the idea that you could pass `in_main_thread=False` as one parameter rather than `sigterm=None, sigusr1=None` as two parameters?

ChristianTremblay, 2017-04-05:
I found the one parameter clearer. But if this is well documented, I see no problem passing None to parameters.

In fact, I'm already doing this in the develop branch.
https://github.com/ChristianTremblay/BAC0/blob/develop/BAC0/scripts/BasicScript.py#L173

JoelBender, 2017-05-01:
This is sitting in stage for the next release.  I even added some documentation :smile:.

closed 2017-05-01 by JoelBender