JoelBender/bacpypes #122: Decoding a proprietary readproperty

Aditya23456, 2017-04-08:
Hi,

I am using bacpypes to read  how many devices are connected behind a bacnet hub device. The vendor gave me 8602 property ID to get connected devices list. I found bacpypes has an in-build translator which translates property name into respective ID. Since proprietary  names can't be mapped I added it into the basetypes.py and mentioned it as readable property of DeviceObject on object.py. While decoding the received message, I get an error at cast_out method saying-- "too many cast components". I see the code supports taglist of only 1 but in debugger I see 20 tags in the response. I am not sure if I did it right by adding proprietary property in deviceobject. Can anyone give any leads? or should I change the code of taglist to support multiple tags?

Thanks,
Aditya

JoelBender, 2017-04-09:
> The vendor gave me 8602 property ID to get connected devices list.
For purposes of the discussion, call this `connectedDevicesList`.  Did they also give you an indication if this was a list or an array?  And are the list/array elements primitive or constructed data types?  If you debug the contents of the tag list that was matched by `Any` and you seeing opening and closing tags, that's a good indication that the `ConnectedDeviceListElement` is a sequence of something(s).
> Since proprietary names can't be mapped...
This is a shortcoming that is being addressed, but its a sticky wicket.  You should be able to subclass any of the "extendable" enumerations and provide additional values, then similar to the object "registration" be able to associate the extended values with the vendor identifier.

This is going to be a substantial change to the encoding and decoding routines.  By subclassing `ObjectType` it is going to effect the encoding and decoding of `ObjectIdentifier` so the vendor "context" has to be carried from the APDU all the way down to primitive data.

Aditya23456, 2017-04-11:
Hi,

I just looked into the docs and vendor mentioned it to be of type "BACnetARRAY Sequence of Unsigned". Since the code is multi-threaded and has callbacks it's hard to follow the sequence to debug. But since I see the data I need to assign correct object type to decode it. I will try to follow what you mentioned. Thanks,
Aditya

JoelBender, 2017-04-11:
Ok, this isn't so complicated as it could have been, `ReadableProperty('connectedDevicesList', ArrayOf(Unsigned))` should do it.