JoelBender/bacpypes #123: Polling all the properties under all objects on local BACnet

SantoKo333, 2017-04-17:
Hi Joe,
I am Santo. I am currently working on a script which needs to poll all the properties under all objects every 5 minutes. I was trying to use RecurrentMultipleReadProperty.py, but it would get stuck for a short time if one point in the list are not valid. So, my way to do this now is to create one prarie dog for each points(ex:('10.10.10.105', 'analogValue', 3, 'presentValue')), all prarie dogs are using the same SimpleApplication. However, this approach somehow doesn't work, I am only getting one properties value from every IP address. (I have multiple objects and properties on same IP address).
I am thinking whether you would know the reason or a better way to achieve my goal.

JoelBender, 2017-04-18:
>  ...it would get stuck for a short time if one point in the list are not valid.

If the point in the configuration list isn't valid the application should have received an error PDU (like "unknown object") almost immediately and the prairie dog should have picked up the next request in the point queue.

> my way to do this now is to create one prairie dog for each points

This won't help much because the application will queue the requests coming down the stack by address and only have one outstanding at a time.  This is for applications that do not have their requests well coordinated (like a web server that is requesting a bunch of points from different devices all at the same time).  There is no way to know how many concurrent requests a particular controller can handle before it starts dropping them and in my experience even if a controller accepts multiple requests, it completes them in a serial order so the total time to satisfy all of them remains about the same.

Your application should behave the same as if you have one prairie dog for each field device, and to give each one its own point list.

> I have multiple objects and properties on same IP address.

If it supported by the controller, try using ReadPropertyMultiple which is more work to built the request and interpret the response, but will give you dramatically better throughput.  Your requests could be as simple as reading the magic `all` property.  Be aware that you could run up against a controller that would like to return everything at once but that would exceed the maximum APDU size of the networking layer and it doesn't support segmentation.

While the work is can be tedious, but you can also walk through a Wireshark capture and line up the requests and responses and see if you can find the bottleneck.  There are some applications in the [bacpypes-pcap](https://github.com/JoelBender/bacpypes-pcap) project that might help.

JoelBender, 2017-04-18:
I stumbled over #124 making sure the project works as expected and it breaks Python3, so that needs to be fixed before the [read property summary filter](https://github.com/JoelBender/bacpypes-pcap/blob/master/ReadPropertySummaryFilter.py) application will work, but it still works for Python2.