JoelBender/bacpypes #127: segmentationNotSupported Error

Aditya23456, 2017-05-05:
Hi,

I have a bacnet wattnode powermeter which has in its manual NO_SEGMENTATION supported. Using bacpypes I can get response for whois broadcast response but using bacpypes I can't make any other query. I get this error-

Device communication aborted: segmentationNotSupported

But sometimes I get through the error and get a response(Very rarely). I looked at earlier issues which suggested from bacpypes 14 this problem was resolved. So I migrated from my 13.2 to latest 16 and made necessary sourcecode changes at my end but still I get same error.

Thanks,
Aditya

JoelBender, 2017-05-06:
You are going to get this error when (1) you are trying to send a request that is too big to send in one piece and your local device object doesn't support sending segmented requests, (2) the remote device doesn't accept segmented requests, (3) the response that the remote device is trying to send back to you needs to be segmented and it doesn't support segmented responses, or (4) the remote device is trying to send you a segmented response and your device object says it doesn't support segmented responses.

Your power meter doesn't support segmentation, so that means you can't send it "big" requests, nor can you expect it to be able to send you "big" responses.  I'm guessing your queries are Read Property requests, which should be fine because that reads one property at time (a very small request) and there are very few property values that would be "big".  If you are using Read Property Multiple, trying to keep traffic down (a good thing) and bundle your reads together to improve performance (also a good thing) you'll have to experiment with how many object/property values can be returned without exceeding the maximum APDU size.

Aditya23456, 2017-05-06:
Hi,

I am actually sending simple read vendor name request.  yes, I am reading one property at a time so both my read request and response are not too big. I tried making other small requests to but have the same error. When I tried reading the model name or vendor name like 2 weeks back it would respond properly and I had segmentation error only when I query for object list. But from yesterday, I have same error for any property request. I changed Basrouter and this time I get a different error saying communicaton aborted:No response from the device. I put my device object with 1024 as Maximum APDU Length Accepted. Should I increase this? Also, I tried communicating with the device from other BACnet software named yet another bacnet explorer and I could read properties from this software in any case( device connected to any router). I am not sure where things are wrong. Please suggest

JoelBender, 2017-05-07:
Please run the application with a debugger attached to the client state machine and send me the output?  My email address is joel{at}carrickbender.com

    $ python samples/ReadProperty.py --debug bacpypes.appservice.ClientSSM

You should see a `DeviceInfoBlock` at the top that looks similar this:

    <bacpypes.app.DeviceInfo object at 0x7fe8c3b50f90>
        address = <Address 10.0.1.21>
        maxApduLengthAccepted = 1024
        segmentationSupported = 'noSegmentation'
        maxNpduLength = 1497

and then below that where it calculates the segment size and the number of segments.  If you see the state machine going through the various states, the `set_state()` calls, then there's something else going on in your application and we'll have to make some other arrangement to look at your application.

JoelBender, 2017-05-10:
Thank you for the log files!  I think this problem is actually with path discovery in the `netservice` module.

JoelBender, 2018-04-07:
There are more updates and tests for segmentation that will be going into the next release, but the changes made for this issue are in stage and ready to be released.  If there are still problems, please create another issue.

JoelBender, 2019-04-12:
Reports that this problem has cropped back up.