JoelBender/bacpypes #128: COVClient.py Bug on value

shorty8100, 2017-05-26:
subscribe 192.168.200.130:61576 1 analogInput 0 presentValue 100
> ('analogInput', 0) changed
    description is <bacpypes.constructeddata.Any instance at 0x03147878>,
    eventState is <bacpypes.constructeddata.Any instance at 0x031478C8>,
    objectIdentifier is <bacpypes.constructeddata.Any instance at 0x0316B120>,
    objectName is <bacpypes.constructeddata.Any instance at 0x0316B148>,
    objectType is <bacpypes.constructeddata.Any instance at 0x0316B170>,
    outOfService is <bacpypes.constructeddata.Any instance at 0x0316B198>,
    presentValue is <bacpypes.constructeddata.Any instance at 0x0316B1C0>,
    reliability is <bacpypes.constructeddata.Any instance at 0x0316B1E8>,
    statusFlags is <bacpypes.constructeddata.Any instance at 0x0316B210>,
    units is <bacpypes.constructeddata.Any instance at 0x0316B238>
('analogInput', 0) changed
    presentValue is <bacpypes.constructeddata.Any instance at 0x03147918>
('analogInput', 0) changed
    presentValue is <bacpypes.constructeddata.Any instance at 0x03147878>
('analogInput', 0) changed
    presentValue is <bacpypes.constructeddata.Any instance at 0x03147918>
('analogInput', 0) changed
    presentValue is <bacpypes.constructeddata.Any instance at 0x03147878>
('analogInput', 0) changed
    presentValue is <bacpypes.constructeddata.Any instance at 0x03147918>

for element in apdu.listOfValues:
element.propertyIdentifier, str(element.value)

as you can see the propertyIdentifier comes correctly but the value does not!
Any suggestions? Thank you

shorty8100, 2017-05-26:
Found the solution, it needs to be processed.
for element in apdu.listOfValues:
	datatype = get_datatype(apdu.monitoredObjectIdentifier[0], element.propertyIdentifier)
	if not datatype:
		raise TypeError("unknown datatype")
	if issubclass(datatype, Array) and (element.propertyArrayIndex is not None):
		if element.propertyArrayIndex == 0:
			value = element.value.cast_out(Unsigned)
		else:
			value = element.value.cast_out(datatype.subtype)
	else:
		value = element.value.cast_out(datatype)

JoelBender, 2017-05-28:
Yep, and that cast_in/cast_out code appears more often than I would like.  Whenever there is a Any, which is a blob of tagged data that needs more interpretation, the "application" needs to provide the additional data.  Congrats on finding it.

closed 2017-05-28 by JoelBender