JoelBender/bacpypes #132: Connecting to Router in other network

SmilingWorlock, 2017-07-31:
Hi,

I am trying to connect to a BACnet device behind a BACnet router behind a router:

My PC is in network A. I only can send (unicast) packages to network B, Adress 192.168.0.4, and only UDP 47808. Adress 192.168.0.4 is a BACnet/IP to BACnet/LONtalk router (device ID 10), which routes to device 20. Broadcasts are dropped in the firewall.

So far, I am only able to use "whois 192.168.0.4" (from the WhoIsIAm.py example) to contact device 10 and get a response.

When using a maintaince acess directly to network 192.168.0.0 I can use Yabe to send out a WhoIs and find both device 10 and device 20, Yabe listing the adress (1 1 via 192.168.0.4) for device 20.

Do you have any idea how to make this work?

Best regards

JoelBender, 2017-08-01:
Your description is missing network numbers, so I'm going to assume that your PC is 192.168.0.3 on BACnet network 2, and your router routes between 192.168.0.4 on network 2 and address 99 on network 1.  Your mission is to send a read request to address 1 on network 1.

If you are running the `ReadProperty.py` application and just say `read 1:1 device 20 objectName` it is going to attempt to find the router to network 1, which you mention is blocked (probably because reading a property uses a confirmed service, and confirmed services shouldn't be broadcast).  There is a `rtn` command which you can use to tell the network layer of your application where to find the router to the destination network, `rtn 192.168.0.4 1` will add that into the local routing tables cache.

After that your read will "magically" become a unicast message on the BACnet/IP network, and then be a unicast message on the LONtalk network.

JoelBender, 2017-08-01:
I'd be curious to see what the output of the `WhoIsRouter.py` sample application if you put in `irt 192.168.0.4` to get the router to dump its routing table, and `wirtn * 1` to send a local broadcast Who-Is-Router-To-Network and see it respond.

SmilingWorlock, 2017-08-04:
Hello,

thank you for your reply. Following you first post I could not get a response from either device. The wirtn command send out an ICMP broadcast which was blocked at the firewall.

irt send out a package and recieved an Ack from device 10, but it did not do anythign else I could see.

I appreciate your help. Unfortunatly I will be out of office for a few weeks. After that I will try your proposed method again and get back to you.

Thank you.