JoelBender/bacpypes #135: ReadPropertyMultiple Unknown Object

JoelBender, 2017-08-29:
The service as it’s currently written throws out the entire request if any of the objects in the read access specifications don’t exist, rather than returning an error in the propertyAccessError part of the ReadAccessResultElementChoice of the ReadAccessResultElement.

The proposed fix is to change `do_ReadPropertyMultipleRequest` function of the `ReadWritePropertyMultipleServices` class where it checks to see if the object exists. Remove the check, guard against referencing `obj._properties.items` since `obj` is None, and `read_property_to_result_element` needs to accept a None for the obj parameter and not bother calling `read_property_to_any`.

dhoomakethu, 2017-08-29:
Referencing the spec just in case required.

### ReadPropertyMultiple Service
### 690 ANSI/ASHRAE Standard 135-2016
###15.7.2 Service Procedure

After verifying the validity of the request, the responding BACnet-user shall attempt to access the specified properties of the specified objects and shall construct a 'List of Read Access Results' in the order specified in the request. If the 'List of Property References' portion of the 'List of Read Access Specifications' parameter contains the property identifier ALL,REQUIRED, or OPTIONAL, then the 'List of Read Access Results' shall be constructed as if each property being returned had been explicitly referenced (see Clause 15.7.3.1.2). While there is no requirement that the request be carried out "atomically," nonetheless the responding BACnet-user shall ensure that all readings are taken in the shortest possible time subject only to higher priority processing. **The request shall continue to be executed until an attempt has been made to access all specified properties. If none of the specified objects is found or if none of the specified properties of the specified objects
can be accessed, either a 'Result(-)' primitive or a Result(+) primitive that returns error codes for all properties shall be issued.**

If any of the specified properties of the specified objects can be accessed, then a 'Result(+)' primitive shall be issued, which
returns all accessed values and error codes for all properties that could not be accessed.

JoelBender, 2017-08-30:
Resolved (limited testing :stuck_out_tongue_closed_eyes:), issue branch merged into `stage` for next release.

closed 2017-08-30 by JoelBender