JoelBender/bacpypes #139: Read property issue regarding value type

Shai-tan, 2017-09-14:
Currently, I implement usage of binary object and especially the "polarity" property.
Registration and instantiation work well but when I try to read the property value an exception occurs:

`
    aValue = wDatatype(aValue)
  File "/usr/local/lib/python2.7/dist-packages/bacpypes/primitivedata.py", line 1088, in __init__
    raise TypeError("invalid constructor datatype")
  TypeError: invalid constructor datatype
`

The datatype is correct `<class 'bacpypes.basetypes.Polarity'>` and the value to convert is ok too.
After investigation, this issue comes from value type.

* `Polarity('reverse')` works
* `Polarity(u'reverse')` fails

I suggest you to replace `str` type in convertion process by `basestring` type than handle `str` and `unicode`.
You can found attached a patch, tested (and validated) on my local server.

[0005_EnumerateUnicode.txt](https://github.com/JoelBender/bacpypes/files/1302013/0005_EnumerateUnicode.txt)

Shai-tan, 2017-09-14:
I suggest another update concerning the object type in unicode format.

* `BinaryInputObject(cfg, ('binary-input', 1), 'objName')` works correctly
*  `BinaryInputObject(cfg, (u'binary-input', 1), 'objName')`  fails when try to read `object-list` of device

In my understanding of python internal processing, the replacement of `str` to `basestring` doesn't change the behavior and is more flexible.

[0005_EnumerateUnicode.txt](https://github.com/JoelBender/bacpypes/files/1302170/0005_EnumerateUnicode.txt)

Then a global replacement of string type can be beneficial for the application

JoelBender, 2017-09-19:
My initial reaction is to decline this and be "strict" because the str <-> unicode conversion has tripped me up more than once.  At some point I included the `isinstance(arg, (str, unicode))` test and forced it to `str(arg)` because I wanted to accept values coming in from other sources (like JSON documents) but make sure `str` was maintained within the library.

With this example and patch limited to enumerations it would seem like the library would be more closely following the [robustness principle](https://en.wikipedia.org/wiki/Robustness_principle) by being more accepting.  The question is, should this be extended to other primitive types as well?  I'm thinking it probably should, and the right way to build confidence is to build in tests.

I'll make a branch and let you know what happens.  Thank you for the contribution.

JoelBender, 2017-09-19:
Check out the branch and run the tests and let me know what you find.

Shai-tan, 2017-09-19:
All tests are ok when I use the branch 139 with my BACnet server.
Nevertheless I found another object concerned by this issue : Address

[0005_AddressUnicode.txt](https://github.com/JoelBender/bacpypes/files/1314165/0005_AddressUnicode.txt)

This object is primitive too so it replies (in my point of view) to your question: Yes all primitive object can be updated according to robustness principle.

JoelBender, 2017-09-19:
Thank you for the update, I included this fix plus tests.

commit d71b7eb9 (2017-10-08): closes #139
closed 2017-10-08 by JoelBender