JoelBender/bacpypes #141: Issue when trying to remove property from object.

Shai-tan, 2017-09-22:
To complete my implementation of `covIncrement` property, I want to:
	* add it when the COV is allowed 
		`self.add_property(ReadableProperty('covIncrement', Real, self._values['covIncrement']))`
	* delete it when it is disallowed.
		`self.delete_property(ReadableProperty('covIncrement', Real))`

In special context of my application, this change can occur during object life and I really don't want to delete object and create it again.

To add the property in my object, I use the `add_property` function from Object class and it works well.
	* The property is present in the `propertyList`
	* The value associated is the one expected.

But when I try to remove it, an issue occurs:
	
`
    self.delete_property(ReadableProperty('covIncrement', Real))
  File "/home/calligp/workspace/P5.4/POC/bacpypes/py27/bacpypes/object.py", line 515, in delete_property
    property_list.remove(prop.identifier)
AttributeError: 'ArrayOfPropertyIdentifier' object has no attribute 'remove'
`

After spending long time to search `ArrayOfPropertyIdentifier` class implementation, I discovered the right one is `Array` class from `constructedata` file.
Effectively this class has no method remove.

I try to add it, see attached patch for details about my test, and now the property is well deleted and no exception raises.
[0006_RemoveProperty.txt](https://github.com/JoelBender/bacpypes/files/1324833/0006_RemoveProperty.txt)

Let me known if my understanding is inexact or if this is really a mistake?
Thanks in advance.

JoelBender, 2017-09-26:
BACnet array's are strange beasts because index 0 is the length of the array, so in addition to removing the element from the values you also have to update the length.  This is already done in the `__delitem__` function, so the remove function could look like this (untested):
```
def remove(self, item):
    indx = self.index(item)
    self.__delitem__(indx)
```
For the `index` function to work the item has to have some way of checking equality which is fine for primitive data types but will not work for `listOfObjectPropertyReferences` properties and other arrays of constructed data types unless we also define a recursive `__eq__` function.  I'm not convinced that this will be worth the effort.

I'll create a branch for this, some tests, and roll it into the next release.

JoelBender, 2017-09-26:
Check out [this commit](https://github.com/JoelBender/bacpypes/commit/6be49d9b156a22460476c7d5c729fd4ff697a778) which has the new function and a sandbox application that shows how I expect the code to work.

Note that I use the `_attr_to_property()` function to grab the property definition from the object so the property definition that is used is the same one that the object defined.  The default definition may have been overridden in a subclass, like the way `presentValue` is overridden in the `RandomAnalogValueObject` definition in various sample applications.

Shai-tan, 2017-09-28:
Indeed I forgot this special case about BACnet array.
Your change works perfectly with my application and now I'm able to add and remove property from my objects.
Thanks !

JoelBender, 2017-10-08:
Good!

commit bd019368 (2017-10-08): closes #141
closed 2017-10-08 by JoelBender