JoelBender/bacpypes #143: Warm restart of bacpypes server

Shai-tan, 2017-10-09:
According to issue #107 already solved, I can restart my BACnet server without restarting my whole application.
For another purpose, I activated the bacpypes internal debug logs and I discovered a very strange log:

```
Traceback (most recent call last):
  File "/home/calligp/workspace/P5.4/POC/bacpypes/py27/bacpypes/core.py", line 151, in run
    asyncore.loop(timeout=delta, count=1)
  File "/usr/lib/python2.7/asyncore.py", line 220, in loop
    poll_fun(timeout, map)
  File "/usr/lib/python2.7/asyncore.py", line 145, in poll
    r, w, e = select.select(r, w, e, timeout)
error: (9, 'Bad file descriptor')
```

This error appears after a warm restart and the file concerned is the broadcast socket from `UDPMultiplexer`.

Symptoms are particularly difficult to understand:
* When client initialize request, the operation is processed and the response is received.
* When server sends a notification (for example COV notification), the client receives nothing.

After many research about socket creation and deletion, I finnaly found the cause of this issue.
It is the class `UDPDirectory` that is derived from `asyncore.dispatcher` which doesn't end communication on the broadcast socket.
This dispatcher must be properly closed using close() function.

See attached patch.

[0007_RestartSocket.txt](https://github.com/JoelBender/bacpypes/files/1369086/0007_RestartSocket.txt)

Maybe same correction can be applied on TCP communication process but I can not test it.

Moreover, I found another minor issue in UDPMultiplexer class regarding broadcasePort:
When the broadcast port is not required, the attribut self.broadcastPort is not initialized (not even to None value).
Then I suggest you to initialize it. It is present in the same patch.

JoelBender, 2017-10-12:
This is mysterious because I expected that when `UDPDirector.close_socket()` was called, which in turn calls `self.socket.close()` that there would be a reentrant call to UDPDirector.handle_close()` which would then call `self.close()` as it does.  But what you are saying with the patch is that telling the director to close the socket isn't enough, that the director itself also needs to be closed.

Would it be sufficient for the `UDPDirector.close_socket()` to look like this (option 1):

```
    def close_socket(self):
        """Close the socket."""
        if _debug: UDPDirector._debug("close_socket")

        self.socket.close()
        self.close()
        self.socket = None
```
Or this might be a little better (option 2):
```
    def close_socket(self):
        """Close the socket."""
        if _debug: UDPDirector._debug("close_socket")

        self.socket.close()
        self.handle_close()
```
I would like to share a test application in the sandbox to see if I can reproduce the behavior and make it part of some testing.  All of the current testing uses virtual networking because I want most of the tests to run without a lot of setup overhead, and my docker skills aren't quite up to the level of spawning a container with a server as the Implementation Under Test (IUT) and spawning another as the Test Device (TD), have them beat on each other, and report the results.

Shai-tan, 2017-10-12:
I agree with you, your correction is more clear because `UDPMulptiplex` doesn't need to know internal implementation (and inheritance) of `UDPDirector` object.

I tested the application restart using the first option and it seems working properly. No error in logs and all response are well transmitted to client.
I think the second is not correct because the function `handle_close` is called by internal process and, in my point of view, it should be kept for this purpose only. Moreover the documentation of `asyncore` module recommends usage of `close` function.
If you look in `asyncore` module implementation, `handle_close` function is just a forward to `close`.

JoelBender, 2017-12-07:
I added this work to the next release, going with option 1.  Creating an application pair for testing is being postponed.

closed 2017-12-07 by JoelBender