JoelBender/bacpypes #144: Instanciation of internal Schedule object

calligp, 2017-10-20:
I am implementing schedule object to comply with B-AAC server standard. But I have some trouble with your implementation regarding two specific properties:
```
@register_object_type
class ScheduleObject(Object):
    objectType = 'schedule'
    properties = \
        [
        ...
        , OptionalProperty('weeklySchedule', ArrayOf(DailySchedule))
        , ReadableProperty('priorityForWriting', Unsigned)
        ...
        ]
```

These properties are not normally writable and it is consistent with basic declaration of schedule object in BACnet specification.
But when the server implements internal schedule process, these two properties become writable according to “K.3.2 BIBB - Scheduling-Internal-B (SCHED-I-B) “

In my first test, I try to override the `properties` array from `ScheduleObject`:
```
class MyScheduleObject(ScheduleObject):
    properties = [
        ...
        WritableProperty('weeklySchedule', ArrayOf(DailySchedule)),
        WritableProperty('priorityForWriting', Unsigned)),
        ...
    ]
```
But it doesn't work because `properties` array is not taken into account. The only solution that I found is to change manually the property configuration:
```
self._attr_to_property('priorityForWriting').mutable = True
```

My first question is to know if my analyze is correct ? Maybe I’m missing something.

If my understanding is okay I have a remark regarding robustness. Modifying the internal object property using "dot chain" is not safe because internal change induces an update in high level application too.
That's why I suggest to replace :
```
self._attr_to_property('priorityForWriting').mutable = True
```
by
```
ScheduleObject.update_property('priorityForWriting', optional = False, mutable = True)
```

See attached patch for details about implementation.
[0008_UpdateBehavior.txt](https://github.com/JoelBender/bacpypes/files/1401759/0008_UpdateBehavior.txt)

JoelBender, 2017-11-02:
The `register_object_type` function call does more than just put a reference to the class into the `registered_object_types` dictionary, it also crawls up through the MRO looking for `properties`.  Your first instinct was correct, creating a subclass of the ScheduleObject.

I created branch for this issue and put a sample application in the sandbox directory.  I would like to avoid changing the class attributes so it is treated similar to a column definition in a database schema where fields don't change from mutable to immutable, for example.  That's a little too magical.

calligp, 2017-12-03:
I agree its magic when it is not explained :)
Now I understand and it works perfectly. 
Thank you !

JoelBender, 2018-02-16:
There's a `LocalScheduleObject` in this `144-schedule-object-instantiation` branch in the samples directory.  I'm planning on putting it in a new `bacpypes.local.schedule` module and moving this to the 0.17 milestone.  Comments here or on the new thread on gitter are welcomed.

JoelBender, 2018-04-07:
All kinds of new code and tests current in stage and ready to be released.

closed 2018-04-07 by JoelBender