JoelBender/bacpypes #147: Unable to write a single element of an array

calligp, 2017-10-31:
I am implementing schedule object to comply with B-AAC server standard. And especially management of weekly schedule
using `WriteProperty` service (K.3.2 BIBB - Scheduling-Internal-B (SCHED-I-B)). In fact I want to write a single element
of the `weeklySchedule` property which has type `ArrayOf(DailySchedule)`.
Below the console logs about write processing.

```
2017-10-27 10:55:01,166 - INFO - BacnetServer - do_WritePropertyRequest - write property weeklySchedule in ('schedule', 0) from 0x0a214a68dbfc
2017-10-27 10:55:01,166 - DEBUG - object - ReadProperty - ReadProperty 'weeklySchedule' arrayIndex=1L
2017-10-27 10:55:01,166 - DEBUG - object - ReadProperty - ReadProperty(weeklySchedule) <Servers.BacnetInterfaces.BacnetObjects.ScheduleBacnetObject.ScheduleBacnetObject object at 0x7f4bc29385d0> arrayIndex=1L
2017-10-27 10:55:01,166 - DEBUG - object - get_datatype - get_datatype 'weeklySchedule'
2017-10-27 10:55:01,166 - DEBUG - object - get_datatype - get_datatype 'weeklySchedule'
2017-10-27 10:55:01,166 - DEBUG - object - WriteProperty - WriteProperty 'weeklySchedule' <bacpypes.basetypes.DailySchedule object at 0x7f4bc08a0ed0> arrayIndex=1L priority=None
2017-10-27 10:55:01,166 - DEBUG - object - WriteProperty - WriteProperty(weeklySchedule) <Servers.BacnetInterfaces.BacnetObjects.ScheduleBacnetObject.ScheduleBacnetObject object at 0x7f4bc29385d0> <bacpypes.basetypes.DailySchedule object at 0x7f4bc08a0ed0> arrayIndex=1L priority=None direct=False
2017-10-27 10:55:01,167 - DEBUG - object - WriteProperty -     - property is not atomic and wrong type
2017-10-27 10:55:01,167 - DEBUG - object - __getattr__ - __getattr__ 'objectIdentifier'
2017-10-27 10:55:01,167 - DEBUG - object - __getattr__ -     - deferring to <bacpypes.object.ObjectIdentifierProperty instance at 0x7f4bc3cf2d88>
2017-10-27 10:55:01,167 - DEBUG - object - ReadProperty - ReadProperty(objectIdentifier) <Servers.BacnetInterfaces.BacnetObjects.ScheduleBacnetObject.ScheduleBacnetObject object at 0x7f4bc29385d0> arrayIndex=None
2017-10-27 10:55:01,167 - ERROR - BacnetServer - _WriteProperty - Error on write weeklySchedule of (u'schedule', 0): weeklySchedule must be of type ArrayOfDailySchedule
```

The write operation is rejected with error ('property', 'invalid-data-type').
Indeed in function `WriteProperty` of `Property` class, the check of input data doesn't take car about array.

That's why I suggest to add a check specific for array with tree condition:
* property is an array
* value to write has same type than array subtype
* array index is not none.

Open question regarding the validation of the array elements to write? Currently, only `Atomic` type has `is_valid`
operation.

Moreover, the capability to write a full array seems not implemented in `Property.WriteProperty` interface. Indeed
when I try to write an array of `DailySchedule` with VTS software on `weeklySchedule` from `Schedule` object, the object
decoded by bacpypes stack is a list of `DailySchedule`. This decoding is ok from my point of view but when the
`WriteProperty` function check if new data is instance of `ArrayOf`, the result is always wrong.

That's why I suggest to handle `list` in exception case for array property and check :
* each element has the right type
* length of list is the expected one (same of previous value).
After that the list can be converted to `ArrayOf` and written.

I have an other question which can (in my point of view) simplify the array management. All array properties
defined in standard have fixed lengths. Only sequence object has dynamic lengths. So the size can be an attribute of
the property. With this information, all write operation can check if the new data has the expected length.

Complementary check regarding write of sequence property, I don't use it for now then I don't know if my solution is
relevant to this case too.

See attached patch for details about proposed implementation.
[0009_WriteArrayIndex.txt](https://github.com/JoelBender/bacpypes/files/1431248/0009_WriteArrayIndex.txt)

JoelBender, 2017-11-02:
This is a good start, thank you.  There are some additional cases in Clause 12.1.5.1 regarding writing to the array length (so the arrayIndex is zero and the value is unsigned) and the type checking should cover that as well.  I created a branch for this issue and I'll start with some unit tests.

JoelBender, 2017-11-08:
You wrote:
> All array properties defined in standard have fixed lengths.
This is incorrect, only those properties like `eventTimeStamps` that have the notation `BACnetARRAY[3]` are fixed length, those like `propertyList` are `BACnetARRAY[N]` and have no fixed length.  There isn't any attribute of a `Property` instance (similar to `optional` and `mutable`) that provides for fixed size arrays, so I'm adding this #149 as an enhancement request.

JoelBender, 2017-11-21:
@calligp Please test this branch with your application.  I think this issue can be closed.

JoelBender, 2017-11-21:
Closing this issue, additional tests in #148, so test that one instead.

closed 2017-11-21 by JoelBender