JoelBender/bacpypes #148: localDate & localTime property are not available in propertyList

calligp, 2017-11-07:
Both properties `localTime` and `localDate` are not present in `propertyList` of my device object which is an instance of `LocalDeviceObject`. That's why I try to set default values during initialization:

```
aConfig['localDate'] = Date(...)
aConfig['localTime'] = Time(...)
```

But the result doesn't work as expected:

```
Traceback (most recent call last):
File "wsapplication.py", line 153, in <module>
  Application(default_args)
File "wsapplication.py", line 71, in __init__
  self._CommunicationServer
File "/home/calligp/workspace/P5.4/HagerEnergyServerMiddleEnd/src/Servers/BacnetInterfaces/BacnetApplication.py", line 39, in __init__
  self.__InitializeServer(aConfiguration)
File "/home/calligp/workspace/P5.4/HagerEnergyServerMiddleEnd/src/Servers/BacnetInterfaces/BacnetApplication.py", line 145, in __InitializeServer
  self._GlobalConfiguration['restartRecipients']
File "/home/calligp/workspace/P5.4/HagerEnergyServerMiddleEnd/src/Servers/BacnetInterfaces/BacnetServer.py", line 128, in __init__
  **aConfig
File "/usr/local/lib/python2.7/dist-packages/bacpypes/service/device.py", line 107, in __init__
  raise RuntimeError("localDate is provided by LocalDeviceObject and cannot be overridden")
RuntimeError: localDate is provided by LocalDeviceObject and cannot be overridden
```

When I look in `LocalDeviceObject` implementation and, in my understanding, these two properties must be added like `objectList` property at the end of the device object initialization.

See attached patch for details.
[0011_localDateTime.txt](https://github.com/JoelBender/bacpypes/files/1450421/0011_localDateTime.txt)

JoelBender, 2017-11-17:
It's actually worse than that, the `objectType` and `propertyList` properties are included in the value of the `propertyList` and they shouldn't be:

> [The Property_List] read-only property is a BACnetARRAY of property identifiers, one property identifier for each property that exists within the object. The Object_Name, Object_Type, Object_Identifier, and Property_List properties are not included in the list.

One of the design concepts of BACpypes object instances is that unless you give it a property value in the kwargs, or that it gets one from the class definition, the property doesn't get a value:
```
>>> from bacpypes.object import Object
>>> o = Object()
>>> [k for k, v in o._values.items() if v is not None]
[]
```
So by that reasoning, this is wrong because the only thing provided by the class definition is the `objectType` and the `propertyList` shouldn't be there:
```
>>> from bacpypes.object import DeviceObject
>>> do = DeviceObject()
>>> [k for k, v in do._values.items() if v is not None]
['objectType', 'propertyList']
```
But when I provide the value as a kwarg I want it to be there and be exactly what I told it, without magic:
```
>>> from bacpypes.constructeddata import ArrayOf
>>> from bacpypes.object import PropertyIdentifier
>>> 
>>> do = DeviceObject(propertyList=ArrayOf(PropertyIdentifier)([]))
>>> do._values['propertyList'].value
[0]
```
With a `LocalDeviceObject` providing some additional property values by default and some automatic handling of those properties, like `localDate` and `localTime`, then it makes sense that the `propertyList` would be like the `objectList` where it is created and magic:
```
>>> from bacpypes.service.device import LocalDeviceObject
>>> ldo = LocalDeviceObject()
>>> [k for k, v in ldo._values.items() if v is not None]
['numberOfApduRetries', 'maxSegmentsAccepted', 'maxApduLengthAccepted', 'objectType', 'apduTimeout', 'apduSegmentTimeout', 'segmentationSupported', 'vendorIdentifier', 'objectList', 'propertyList']
>>> ldo._values['propertyList'].value
[10, 'propertyList', 'numberOfApduRetries', 'maxSegmentsAccepted', 'apduTimeout', 'apduSegmentTimeout', 'segmentationSupported', 'vendorIdentifier', 'maxApduLengthAccepted', 'objectType', 'objectList']
```
And your patch helps a little because it does appear, but only when the `objectList` is not provided, which seems wrong.  If the `objectList` is supposed to be magic when the kwarg is not provided, then maybe the `propertyList` should act the same way - _or_ - both the `objectList` and `propertyList` are going to be like `localDate` and `localTime` where they will always be magic and init values cannot be provided.

In this branch I removed the `propertyList` management from the `Object` class and moved it to a new `LocalObject` class, then stuffed it in the inheritance chain of the `LocalDeviceObject`.  I also "promoted" the `objectList` property to be something managed by the `LocalDeviceObject`.  In order for the `objectList` to be created with the correct initial value (a single element array containing the object identifier of the device object) then the `objectIdentifier` property is required.

This is a pretty significant change in behavior that might break backward compatibility if there are applications that are relying on `propertyList` to be automatically handled for generic objects like analog values.  The applications can do this to get the behavior back:
```
class LocalAnalogValueObject(LocalObject, AnalogValueObject):
    pass
```
In the grand scheme of things I think this is a better separation of responsibility.

JoelBender, 2017-11-17:
If the `propertyList` is managed by the new `LocalObject` class, and when a property value is **None** it means the object doesn't have that property, then when the value is changed to None it should be removed from the list and when it's not None added to the list if it's not already there.

JoelBender, 2017-11-19:
Thinking more about the comment above, I wrote a WriteProperty function in LocalObject that overrides the one in Object and changes the property list accordingly, but I think that will negatively impact performance for the vast majority of the time that property values are simply changing, and not being added or deleted.  Now I'm on the trail of making a read-only version of the property list that returns the correct value on demand, like the way the current date and time properties do for the device object.

JoelBender, 2017-11-19:
Ah, much nicer.  Passes all my "no barf" tests, but no formal tests written yet.  There is a BACpypes developer out there that is calling `add_property` and `delete_property` to make it show up and disappear from the property list, this improvement makes that obsolete.

calligp, 2017-11-29:
Indeed you found a bug and you corrected it with the definition of CurrentPropertyList property. For me it works well, 'objectType' (and co.) are no more present in the `propertyList`.

In my understanding the CurrentPropertyList property only checks the content of existing `propertyList` but doesn't add missing ones.

That's why it doesn't change my issue regarding the presence of 'localTime' and 'localDate' properties in the `propertyList` of my device object.

JoelBender, 2017-12-01:
Ah, found the bug.  If I give the property a default value (and I use an empty tuple in the latest commit) then it's not None and therefore exists:
```
>>> from bacpypes.service.device import LocalDeviceObject
>>> ldo = LocalDeviceObject(objectIdentifier=('device', 1), vendorIdentifier=999)
>>> ldo.propertyList.value
[10, 'apduSegmentTimeout', 'apduTimeout', 'localDate', 'localTime', 'maxApduLengthAccepted', 'maxSegmentsAccepted', 'numberOfApduRetries', 'objectList', 'segmentationSupported', 'vendorIdentifier']
```

calligp, 2017-12-03:
Now it is correct (and complete) for me. In fact the correction of the initial issue is pretty simple but really well hidden.
Thank you

closed 2017-12-07 by JoelBender