JoelBender/bacpypes #151: Event Enrollment: Fault notification

calligp, 2017-12-04:
As described in BACnet specification, the event enrollment is linked to the reliability of the object reference reliability. For this purpose a specific notification parameter is defined but not implemented in your library, it is named "ChangeOfReliability" with identifier 19.

I suggest you the patch attached. [0012_ChangeOfReliabilityParameters.txt](https://github.com/JoelBender/bacpypes/files/1527420/0012_ChangeOfReliabilityParameters.txt)

Another minor point regarding the event enrollment object implementation. The property `eventDetectionEnable` is required and not optional.

JoelBender, 2017-12-05:
Thank you for the patch, I committed it to `stage`.  I'll be back to this issue for the `eventDetectionEnable` part before I close it.

JoelBender, 2018-04-07:
This kept getting pushed back because it never got assigned to a milestone and I lost track of it.  A thousands pardons!

closed 2018-05-04 by JoelBender