JoelBender/bacpypes #152: COVServer.py sample does not update Subscriber of a status change asynchronously

Walleye007, 2017-12-07:
I have run (--debug and --console) the sample and tested it using YABE. I can query the covserver and all is okay. Exiting works. When I add a subscription to "bv" and try to exit, the application hangs until I send it a message or query by YABE. If I change the value of bv using "set" and then "trigger", I see that the cov message back to YABE is not sent until another message from YABE sending a subscription update is received. Once that happens, covserver will send the cov message and the regular interactions. If I trigger it twice, then two messages are delayed until the YABE updates the subscription. Why does this sample not function as expected? I am using Wireshark to monitor the communications to verify what I am seeing in YABE. I am using Python 2.7.9.

Thank you!

JoelBender, 2017-12-08:
After you add the subscription, and I assume you are using YABE as the client, when you enter `status` on the server console do you see the subscription?  For example, when I'm running the server on port 47808 and the client on 47809:
```
> status
('binaryValue', 1) <bacpypes.object.BinaryValueObject object at 0x7f1ca8f2d790>
    10.0.1.21:47809 proc_id=1 confirmed=True lifetime=30
```
And does it do the same thing if you are running the server without `--console`?  It seems like it might be the console thread starving the main thread.  If you run with `--debug bacpypes.consolecmd`` then you should see `do_exit()` called and the thread exits.

Walleye007, 2017-12-08:
YABE is the client. YABE queries COVServer. Then I add a subscription to "bv" object. Once this happens, I cannot exit COVServer without another network message. I have included a capture of the debug below. If I do not use the console command, I have the same issue when I exit with a ctrl-c. If I use YABE to subscribe, then I can no longer exit with ctrl-c until I send another BACNET UDP message of some sort to it. 

See attached network capture, YABE screen and debug output. 
Events:
12:30:03 - YABE queried COVServer discovering objects
12:30:07 - YABE COV subscribed to COVServer "bv" object 
12:30:30 - set bv.presentValue = 1
12:31:53 - YABE automatically renews subscription - and COV notification is sent back first and then confirmation of subscription - but the COV notification should have been sent at 12:30:30!
12:32:30 - exited COVServer 
12:33:53 - YABE sends subscription renewal and COVServer now shuts down displaying "DEBUG:__main__:fini

[image]
[image]
[traceCOVSendJB.txt](https://github.com/JoelBender/bacpypes/files/1543457/traceCOVSendJB.txt)

JoelBender, 2017-12-08:
I'm going to try and reproduce this, in the mean time maybe there's other folks that can help.

Walleye007, 2017-12-08:
core.py is hanging at the asynccore.loop() function call in the run() function. I also tried to set the method to poll with the same result. Once an outgoing packet is in the queue, it does not return from the call until a network message is received.

Walleye007, 2017-12-08:
The behavior is the same with Python 3.6.

Walleye007, 2017-12-09:
In core.py, when taskManager.get_next_task() is called, it returns a value of 120 seconds for the delta. Then asyncore.loop() is called with this 120-0.001=119.999 seconds. Because of this, the loop will not return for a very long time, making it not exit the program. For now, I have set delta=0.001 before the call to asyncore.loop() and COVServer.py now behaves properly. 

What code changes should be made to core.py to correct this issue? 
  
core.py snippet:
 running = True
    while running:
#       if _debug: run._debug("    - time: %r", time.time())
        loopCount += 1

        # get the next task
        task, delta = taskManager.get_next_task()

JoelBender, 2017-12-11:
The value that is being returned is the amount of time that the stack expects to be able to wait until it should break out of the `select()` call inside the `run()` function and do something else that is scheduled as a "task".  The actual delay passed to `select()` is the `delta`.  There is a variable called `sleeptime` which is passed to the `time.sleep()` function, the side effect is releasing the interpreter to give time to other threads.  If you call `enable_sleeping()` before calling `run()` then every time through the run loop other threads can get time.

So try a couple things; first, try calling `enable_sleeping()` before `run()` and see if the run function is starving another thread, but it also seems likely that the flip side is happening, i.e., the other thread is starving `run()`.  You can also try passing a very small value to the `spin` parameter of run which will shrink the amount of time the thread stays inside the `select()` call.

@ChristianTremblay has gone down this path (join us over on [gitter](https://gitter.im/JoelBender/bacpypes) and can confirm that he is seeing similar behavior.  I'm not sure how to proceed yet.

JoelBender, 2017-12-22:
I think the bottom line at this point is that Windows applications should pass a very small `spin` value to `run()` until there is some other way to signal between threads, and that [pyuv](https://github.com/saghul/pyuv) might be an option.  The alternative is to write non-threaded applications.  I'm merging this (with the delta value clipped) and closing the issue.

commit d826f1e8 (2017-12-22): closes #152, with future work needed to help Windows mutli-threaded applications
closed 2017-12-22 by JoelBender