JoelBender/bacpypes #155: Property list is no more available for most of object types.

calligp, 2017-12-15:
When I try to read the property list for different kind of object on my BACnet server I receive an error `unknownProperty`. The only exception is the device object which has the correct content.

My instinct tells me it is linked to the correction of the issue #148 where you processed a refactor of `propertyList` using the `CurrentPropertyList` object. That's why I try to add this property in `Object` global class replacing standard definition of `propertyList` property.

See my patch attached for more details.
[0013_PropertyList.txt](https://github.com/JoelBender/bacpypes/files/1562198/0013_PropertyList.txt)

This patch corrects the issue for some types like `notificationClass`, `eventEnrollment` but not for `binary` and `analog` set of objects.

Let me know if my understanding is correct and I will try to find out why analog and binary objects are rebellious.

calligp, 2017-12-15:
The last part of this issue regarding `analog` and `binary` objects is caused by an issue in my specific implementation. So, from my point of view, the patch attached corrects the presence of `propertyList` for all object types.

[0013-3_PropertyList.txt](https://github.com/JoelBender/bacpypes/files/1563404/0013-3_PropertyList.txt)

JoelBender, 2017-12-17:
I would like to keep amount of auto-magic code in the basic object descriptions to a minimum, so if there additional features to be added then make a derived type and add them there.  For example, the `DeviceObject` knows about the existence of the local date and time, but the `LocalDeviceObject` is a derived version that provides automatic functionality.  Similarly, the `AnalogValueObject` has the standard properties but no default values or behavior, but the `RandomAnalogValueObject` in the sample code adds some.  If you have objects in your server application that you are building, make a derived class and include `CurrentPropertyListMixIn` to pick up the additional feature.  You will have to do this anyway if you want to eventually make your analog value objects has a writable present value since the standard only requires "readable."

I'm going to be doing the same thing with the schedule object I'm working on, the `LocalScheduleObject` will be put in a `bacpypes.service.sched` module and nothing will change with the `ScheduleObject` in the `bacpypes.object` module.

calligp, 2017-12-22:
Oh okay your solution is good. Sorry for my misunderstanding.

closed 2017-12-22 by calligp