JoelBender/bacpypes #158: Receiving event notifications

cemremengu, 2017-12-22:
Any example for receiving event notifications on a local device application we created? 

I will basically assign the notification class in my (physical) controller to push the event notifications to my device.

Any help or examples are appreciated, and thanks for the great work!

JoelBender, 2017-12-22:
The closest thing I have now is the `COVClient.py` sample application that has methods for processing confirmed and unconfirmed COV notifications.  Event notifications are very similar, and both APDU's share the same notification parameters.  The fact that there's a huge list of choices makes for a lot of code to check, e.g., if `apdu.eventValues` is not None then the event has values, and if `apdu.eventValues.changeOfBitstring` is not None then it was the choice, so `apdu.eventValues.changeOfBitstring.referencedBitstring` will have a value, as will `apdu.eventValues.changeOfBitstring.statusFlags`.

cemremengu, 2018-04-08:
My apologies. Resolved, thank you for the help.

closed 2018-04-08 by cemremengu