JoelBender/bacpypes #160: Issue with LocalDeviceObject in most of the sample code

erincamille, 2018-01-03:
Hi there,

In most of the sample code, local devices are initialized like so:

```
this_device = LocalDeviceObject(
  ... 
  objectIdentifier=int(args.ini.objectidentifier),
  ... 
 )
```
I tried to "find" these devices using a BACnet explorer, and they were all showing up as Analog Input Objects. In order for the explorer to correctly identify these objects as devices, I had to change the objectIdentifier line to

`objectIdentifier=('device', int(args.ini.objectidentifier)),`

as it appears when initialising other objects. Not sure if this is a mistake or intentional. Thanks for your help.

JoelBender, 2018-01-04:
I'm not sure why this is happening for you, I can't reproduce it.   When you create an object and give it an object identifier that is an int or a long and not a tuple, it should add the object type as a convenience:
```
>>> from bacpypes.object import AnalogValueObject, DeviceObject
>>> avo = AnalogValueObject(objectIdentifier=12)
>>> avo.debug_contents()
    _app = None
    objectIdentifier = ('analogValue', 12)
>>> do = DeviceObject(objectIdentifier=13)
>>> do.objectIdentifier
('device', 13)
```
LocalDeviceObject is a subclass of DeviceObject and has additional configuration checking and default values and such, but should behave the same way:
```
>>> ldo = LocalDeviceObject(objectIdentifier=12, vendorIdentifier=999)
>>> ldo.debug_contents()
    _app = None
    objectIdentifier = ('device', 12)
    vendorIdentifier = 999
    objectList
        length = 1
        [1] = 12
    maxApduLengthAccepted = 1024
    segmentationSupported = 'segmentedBoth'
    localTime = ()
    localDate = ()
    apduSegmentTimeout = 5000
    apduTimeout = 3000
    numberOfApduRetries = 3
    maxSegmentsAccepted = 16
```

JoelBender, 2018-01-04:
**Ahhh!**  I see it.  The `objectList` property should contain object identifiers and as shown in the sample above it just has `12` and not `('device', 12)` like it should.  Interesting bug, thank you for finding it!

JoelBender, 2018-02-16:
The `ObjectIdentifierProperty` has some error checking and these "magical" lines:
```
        elif isinstance(value, (int, long)):
            value = (obj.objectType, value)
```
This coerces an int/long into the correct kind of tuple.  I guess this is a feature, it seemed like a good idea at the time (I'm not so convinced anymore).  When the `LocalDeviceObject` is constructed, the code references `kwargs['objectIdentifier']` but at that point the value hasn't been coerced to anything, so you're getting the behavior you've been seeing.

There are three options:
1. Remove the magical code and change all of the applications to match the code that works that you found, or
2. Duplicate the coercion so the object identifier keyword argument is "fixed", or
3. Pick up the coerced object identifier after the `super(LocalDeviceObject, self)` call and use that value building the object list.

Option (1) seems like a pretty big change for a point release.  I would make this kind of change for 0.17 and raise a deprecation warning, then make it a type error in 0.18.

Option (2) is simple and a good stop-gap fix.

Option (3) is ugly.

JoelBender, 2018-02-16:
I think the `160-local-device-objid` branch I just created fixes the problem, please test!

erincamille, 2018-02-16:
Great, thank you!

closed 2018-02-17 by JoelBender