JoelBender/bacpypes #161: BACpyes application for BBMD + Router

sangeeths, 2018-01-04:
Hi all,

I'm looking for a BACpypes stack that acts as both Router and BBMD. The Router could be any IP-to-X [1] Router where the IP side of the Router should also acts as a BBMD i.e. both Router and BBMD applications should bind to the same IP address.

I have referred the sample programs for BBMD [2] and IP2IPRouter [3]. They work fine when running separately binding to two different IP address from the same subnet. But I couldn't find any samples that highlights BBMD and BIPSimple in one stack. Are they meant to co-exist and bind to same IP address?

If we make UDPMultiplexer sharable [4] between BIPBBMD and BIPSimple applications, then we could make both of them bind to the same IP address. But the downside is all the broadcast between BIPBBMD and BIPSimple will be dropped by the UDPMultiplexer [5]. So broadcast communication from BBMD will not reach BIP and vice versa.

```
ip_address = "192.168.1.10/24:47808"

bip = BIPSimple()
bip_annexj = AnnexJCodec()

bbmd = BIPBBMD(ip_address)
bbmd_annexj = AnnexJCodec()

mux = UDPMultiplexer(ip_address)

bind(bip, bip_annexj, mux.annexJ)
bind(bbmd, bbmd_annexj, mix.annexJ)
```

I'm looking for suggestions on how to make an application that has both BIPBBMD and BIPSimple binding to same IP address. 

Thank you,

Sangeeth

[1] X could be IP, MSTP, Ethernet, etc.
[2] https://github.com/JoelBender/bacpypes/blob/master/samples/BBMD.py
[3] https://github.com/JoelBender/bacpypes/blob/master/samples/IP2IPRouter.py
[4] We could make reuse=True in UDPDirector - https://github.com/JoelBender/bacpypes/blob/master/py27/bacpypes/bvllservice.py#L96 and that will allow to re-use the socket. 
[4-1] Also we could make UDPMultiplexer's annexJ (server) to deal with more than one serverPeer. Say, we can support a list of clients to bind to. In this case serverPeer could be [bip_annexj, bbmd_annexj]
[5] https://github.com/JoelBender/bacpypes/blob/master/py27/bacpypes/bvllservice.py#L144-L147

JoelBender, 2018-01-05:
_Routers_ are things that manage traffic between two _networks_, which in the case of BACnet/IP includes the combination of both the IP network (e.g., 192.168.1.0/24) and the port (e.g., 47808).  So if you want an application that is a BBMD on one network and a non-BBMD (a.k.a. simple) on another network with the same IP address, you need to give them separate ports (e.g., 47809).

The `BIPSimple` and `BIPBBMD` instances live at exactly the same spot in the stack of objects, just above the UDP layer and just below the network layer.  The only difference in functionality is what happens with broadcast messages.  Most of the functions of a BBMD (forwarding broadcasts and maintaining foreign device registration) never reaches the network layer unless it should.  A BIPBBMD with an empty BDT and foreign registration disabled is essentially a BIPSimple.

If you want to pursue this, for all incoming (upstream) packets from the socket you would have to decide which of the two pieces would get the packet, and then figure out how to splice the packets continuing upstream to the network layer.  There is a bit of code in the UDPMultiplexer that is for making the distinction between Annex H and Annex J traffic, splitting it on the way up the stack (assuming it's going to go to two different network/application stacks), and merging it on the way down.

BACpypes doesn't support MSTP (but there are commercial USB to MS/TP dongles that are a possibility) and it doesn't support Ethernet without using "raw sockets" and/or mucking around with pcap libraries and such.

JoelBender, 2018-01-05:
Oh, and it just occurred to me that maybe you are looking for something that adds an application layer on top, maybe call it a BBMD2IPRouterApplication or something.  For that you need to add in the code at the top of the `BIPSimpleApplication.__init__()` which does all the application service and state machine access point stuff.

ChristianTremblay, 2018-01-09:
The hard coded timing required by MSTP makes it reeeaaaalllyyy difficult for python.....